flipperdevices/flipperzero-firmware #2021: SubGhz remote layout

Jakesta13, 2022-11-15:
### Description of the feature you're suggesting.

I found that I use the SubGhz mode mostly for remotes, though it gets extremely disorganized really quickly.

Making the SubGhz able to be more organized like the Infrared app does for remotes would make this feature more usable, though it doesn't have to be the same (E.g perhaps create a folder on request as to group all the buttons for one remote for example).

Currently the only way to organize them is via computer and manually making the folders and moving the recorded files per folder.

Skorpionm, 2022-11-16:
how do you suggest sorting the keys?  completely different codes can be recorded on one remote control, while for rolling they generally differ greatly. if you just catch the air, it’s not at all clear what and where to fly from

Jakesta13, 2022-11-16:
> how do you suggest sorting the keys? completely different codes can be recorded on one remote control, while for rolling they generally differ greatly. if you just catch the air, it’s not at all clear what and where to fly from

I guess for rolling codes, have an option to not save into a specified "remote" - I haven't recorded such codes so I'm not sure if I fully understand what you mean.

For keys, I don't think I understand what those are referring to? 

I just imagine being able to organize them in some way would make it much more user friendly overall - and for those oddball codes let the user choose not to organize them into a remote I guess would work too

jrosn, 2022-11-19:
+1

I use the Flipper to control a lights in my apartment.
It would be more convenient for me if I could organize a keys logically related to each other

// sorry for my English

WildRat, 2022-12-15:
+1 Many Sub-GHz remotes have couple buttons on them and send different commands for same device. It be very convenient if we can have them represented as buttons on screen, and not as separate files (or combine different remotes in one). Like it's done with IR remotes functionality of Flipper.

Skorpionm, 2024-04-16:
made in third party applications

closed 2024-04-16 by Skorpionm