flipperdevices/flipperzero-firmware #2022: U2F on Android not working

LowSkillDeveloper, 2022-11-15:
### Describe the bug.

I can't use U2F on my android smartphone. But everything works on the computer. 
Doesn't work for everyone or just me?

[image]
[image]
[image]

### Reproduction

1. Go to google accaunt settings
2. Security settings - two-factor verification - security key
3. Launching authorization through the key.
4. Сhoose USB
5. Insert the key. 
6. Flipper shows connected.
7. A confirmation message pops up, click ok.

And that's it, nothing happens.

### Anything else?

Samsung Galaxy S20 (Android 13) official firmware

LowSkillDeveloper, 2022-12-03:
Now it works. Looks like the problem is solved in https://github.com/flipperdevices/flipperzero-firmware/pull/2073.

closed 2022-12-03 by LowSkillDeveloper

cristian3602, 2024-09-05:
Hello Everyone, I'm just having the same behavior described on the original comment. I'm trying  to use the U2F with a Samsung S23.  The flipper firmware version is 0.105.0