flipperdevices/flipperzero-firmware #2023: Add a Keypad Lock mode to dummy mode in GUI

simone7121, 2022-11-16:
### Describe the enhancement you're suggesting.

It would be nice if it were possible to implement the Keypad Lock functionality evenon the dummy mode, so that if a person has possession of your Flipper Zero, to play on it, it doesn't have the possibility to enter into the brainac mode for pentest purpose.

GMMan, 2022-11-23:
Anything stopping you from just locking the whole device with PIN if that's a concern?

simone7121, 2022-11-23:
> Anything stopping you from just locking the whole device with PIN if that's a concern?

It's already a valid choice, but it would have been even better to lock it in dummy mode, so that the real owner only would use the brainac mode, and other persons can only play with the games  installed. But if this lock is unnecessary for you, they can insert a selector in the settings menu to enable or disable this feature. 😉

skotopes, 2023-06-22:
Yep

closed 2023-06-22 by skotopes