flipperdevices/flipperzero-firmware #2026: TX Issues On Supported Band

ferrellw, 2022-11-18:
### Describe the bug.

I am having issues getting my Flipper to transmit on 433.92. Using a wireless fan [remote ](https://fcc.report/FCC-ID/A25-TX045/4712003)I can receive data but when I try to send it, it states "Transmission is blocked. Transmission on this frequency is restricted in your region."

Reading the official documentation the frequency (433.92) is allowed for my locale (US) 433.05 - 434.79 MHz.

Using an unofficial firmware, I can again, read the data but when I send the code it states "Transmission is blocked. This frequency is outside of the default range.

It only appears on the Sub-GHz "Read" option, not "Read RAW". Transmit on "Read RAW" works fine.

What am I missing?

https://imgur.com/a/BaCnkcH

### Reproduction

1. Flash with official firmware (0.71.1) and locale (US).
2. Enter Sub-GHz menu.
3. "Read" button press from wireless remote.
4. Select captured keypress and "Send" it.

### Target

Wireless Fan Remote @ 433.92

ferrellw, 2022-11-19:
One observation is that the capture is being decoded as a gate controller type device (see images; GateTX 32bit).

skotopes, 2023-03-13:
Please reopen if you still have this issues with information I've requested.

closed 2023-03-13 by skotopes