flipperdevices/flipperzero-firmware #2027: Bad USB Demo gets stuck on every Windows machine I tried so far

dnns92, 2022-11-19:
### Describe the bug.

As the title suggestes, the bad_usb/demo_windows app freezes the flipper. It either does not recognize the PC at all or freezes at 3%. 

The microsoft systems I tried this on are 
21H2 Build 2200.1219 (stuck at 3%)
21H2 Build 2200.1098 (doesnt even recognize the machine)

The bug is not recoverable and can only be fixed from a user side by force-powering the flipper down.

### Reproduction

1. Switch on
2. Go to bad_usb/demo_windows 
3. connect it to a running windows machine and start
4. it either doesnt even recognize that it has been connected or gets stuck at 3 %

### Anything else?

I have to say the software on the device has been quite a dissapointment so far. I couldn't clone a single debit or credit card of the 4 of mine that I tried nor do the usb functions work, I was able to clone some NFC Cards, but no emulation ever worked.

skotopes, 2022-11-19:
Hi, there is known issue in BadUSB that is going to be fixed in next release.

skotopes, 2022-12-01:
Should be fixed in latest release. Please reopen if issue still persist.

closed 2022-12-01 by skotopes