flipperdevices/flipperzero-firmware #2031: Don't attach the device name in all BLE advertising packages

Semper-Viventem, 2022-11-20:
### Describe the enhancement you're suggesting.

Currently, the BLE in the flipper is constantly broadcasting the static address and device name (including the unique flipper's name) to the BLE ether in all advertising packages. This is not a vulnerability but makes a risk of snooping on device owners. For example, any BLE-sniffer botnet in a city can track the movement of devices and compromise the owner's home address.

There are two things that will help reduce the risk:
* Add device name only in the pairing mode.
* Use a `Random Private Resolvable Address` for BLE packages.

For example, all apple devices change BLE addresses every 15 minutes to avoid the risks of surveillance.

UberGuidoZ, 2022-11-23:
There are alternative firmware options that have addressed this issue. Might be able to take some thoughts from them?

UberGuidoZ, 2022-12-04:
Hopefully! Would be a great addition.

Az-Neter, 2022-12-04:
> There are alternative firmware options that have addressed this issue. Might be able to take some thoughts from them?

'Install CFW' should not be the first answer to issue requests for the official firmware. 
Also as for the validity of your statement, you literally seem to be guessing.
Does YOUR cfw have this fix, because I'm pretty sure we're on the same release right now, and mine sure doesn't seem to.

If this already exists in one of the only two notable cfws, by all means it should be copied into official repos, but I'm pretty sure the ofw devs can do a little better than simply copying a shoddy cfw patch.
Considering OP offered two potential fixes, there is no need to even recommend software that is explicitly unsupported by the devs of this repo.

UberGuidoZ, 2022-12-04:
They provided answer was NOT install custom firmware, merely a comment, stating this solution exists if one wanted to seek it out. I am not able to post a link to it as official has made it very clear they do not want links to certain firmware‘s.

Yes, the one I am running does have that option. It is not an assumption. There are instructions on how to do it right on the README of the firmware.

Unfortunately, the official firmware developers have decided to block a number of the custom firmware developers from Discord and GitHub so their changes are no longer being provided, even if it was a change that would be welcomed by Official. I’m also not going to link it, nor DM somebody information, per their request long ago.

At no point did I recommend installing custom firmware. Please read the comment again. I suggested looking at them for ideas if desired. I am quite respectful of the rules and requests by the official firmware devs/staff and certainly believe your comments are unnecessary. If they have any issues with the comments, they are more than welcome to remove it. Take care.

skotopes, 2022-12-05:
We'll check what we can do in next iteration over ble subsystem.

Semper-Viventem, 2022-12-26:
Here is a small illustration of this issue, I just have found a few flipper devices walking around the city.

[image]

[image]

UberGuidoZ, 2023-01-02:
> @UberGuidoZ did they?

Sorry, didn’t see this before - yep, Nano did! They addressed the BLE name, BadUSB name, and more with the thought of privacy.

abcde-r, 2023-04-12:
[WiGLE](https://wigle.net/mapsearch?maplat=43.505239756108296&maplon=-50.85947064476461&mapzoom=3#bluetoothSearch) was historically a WiFi scanning application and website database. They added Bluetooth scanning at some point. This means that Flippers are showing up in their crowdsourced maps.

Here's an example map of US and European Flipper distribution, thanks to the choice to broadcast unique names.
[image]

I agree with the sentiment of this issue, that a security tool shouldn't be _defaulting_ to settings that make its customers trackable.

skotopes, 2023-04-13:
I'll bring this topic to internal discussion, we'll redesign this part

abcde-r, 2023-04-13:
Thanks for pushing it forward. Keep in mind that you need to do what the original posted said, and use a "Random Private Resolvable Address" that rotates every ~15 min or so (common industry best practice). If you continue to use a "Public" address (or even a "Random Static" address (if it doesn't rotate every boot)), then the devices will continue to be uniquely identifiable and trackable.

Semper-Viventem, 2023-04-13:
@abcde-r right. But for the first iteration, I guess, it would be enough to hide a device name at least. Even this will greatly reduce the risk of tracking.

UberGuidoZ, 2023-04-17:
Hiding the device name may make it difficult to connect to the correct one for those that have multiple devices. Having a name that does not link directly to the device itself, that also rotated, but is viewable through the Bluetooth menu, would be the best of all worlds.

Semper-Viventem, 2023-04-17:
> Hiding the device name may make it difficult to connect to the correct one for those that have multiple devices.

If device is pared it would be easy to connect using a known address. If a device is not paired, the name may be visible in the pairing mode.

UberGuidoZ, 2023-04-20:
Yes - as long as the name only shows up in Pairing mode. But if the name is changed (randomized) it may be difficult to know the exact one unless it is displayed on the Flipper itself during the process.

turbocool3r, 2023-10-08:
> Yes - as long as the name only shows up in Pairing mode. But if the name is changed (randomized) it may be difficult to know the exact one unless it is displayed on the Flipper itself during the process.

Perhaps it would be a nice compromise to allow to prevent devices from being advertised? Typically phones and laptops/PCs, wireless speakers are only discoverable when you explicitly put them into the discoverable mode (by pressing/holding a button or entering bluetooth settings), otherwise you can only connect to them when your device has already been paired. BLE devices do not advertise themselves in this mode, but since the app tries to connect to the device anyway this shouldn't be an issue. That way keeping the name is OK as well since it is only broadcasted for a short period of time while a pair is being created.

UberGuidoZ, 2024-01-02:
Agreed - excellent solution.

skotopes, 2024-01-02:
We'll integrate new private address mode and hide name from announce. It will take couple releases before it arrive.