flipperdevices/flipperzero-firmware #2034: BT remote Keynote vertical layout

Kami-no, 2022-11-21:
### Description of the feature you're suggesting.

It would be great to have at least an option to switch Keynote button layout to vertical with buttons at the top. Horizontal button layout is not convenient if you're the speaker.

Smooklu, 2022-11-23:
Hmm, would it make sense to have it be switchable or just have it always be a vertical layout?

Smooklu, 2022-11-23:
If it's always gonna be vertical, I can should be able to easily do that.

Kami-no, 2022-11-23:
For me there is no point to have it horizontal.

skotopes, 2022-12-01:
Make sense, but it's going to be difficult to make nice UI/UX for application in general

Kami-no, 2023-03-02:
May be another copy of app with vertical layout?

skotopes, 2023-06-22:
possibly

Kami-no, 2023-06-22:
PR #2794

skotopes, 2023-10-12:
Fixed in https://github.com/flipperdevices/flipperzero-firmware/pull/2794

closed 2023-10-12 by skotopes