flipperdevices/flipperzero-firmware #2036: BadUSB: ALTSTRING is not working

undermuz, 2022-11-21:
### Describe the bug.

Using demo_windows script i am getting this:
```
Hello World!
Hello World!
Hello World!
•̐򿛨
==================================
```
Instead of this:
```
Hello World!
Hello World!
Hello World!
This line was print using Alt+Numpad input method. It works even if non-US keyboard layout is selected
==================================
```
OS:
```
Edition: Windows 11
Version: 21H2
Build: 22000.1219
```
Flipper:
```
FW: 0.71.1 [14-11-2022]
HW: 12.F7B9C6
```

### Reproduction

1. Open BadUSB
2. Select demo_windows
3. Connect to the PC
4. Click Run

skotopes, 2022-12-01:
Looks like num lock off. @nminaylov can you check?

Ethanol0001, 2023-03-03:
one other thing I found to mess with badUSB is having caps lock on. Try making sure both numlock and capslock is off

closed 2023-06-22 by skotopes