flipperdevices/flipperzero-firmware #2040: Mifare DESFire UID emulation not possible due to "Can not parse file"

mh-, 2022-11-22:
### Describe the bug.

This is related to #1730, but I want to clarify that I do not expect the encrypted (and thus unreadable) data to be saved. 
What I do want is that the UID is saved and can be emulated later, by opening a "saved" card.

I only have one particular DESFire card, and it is possible to use a cloned version of that card in its system, by just emulating the UID. I know this works, because the Android app "NFC Card Emulator Pro (Root)" can do that.
Of course, only the UID can be cloned, no encrypted data, but the reader accepts it anyway.

Now, after reading this card, the data which is saved by the firmware  is 'a bit short', because of these problems
```
7618 [I][NfcWorker] Mifare DESFire detected
7832 [W][MifareDESFire] Bad DESFire GET_KEY_SETTINGS response
7842 [W][MifareDESFire] Bad DESFire GET_KEY_SETTINGS response
```
so the file ends after the `Application IDs:`
```
Filetype: Flipper NFC device
Version: 2
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic
Device type: Mifare DESFire
# UID, ATQA and SAK are common for all formats
UID: 04 1B 2C EA 6B 3B 80
ATQA: 44 03
SAK: 20
# Mifare DESFire specific data
PICC Version: 04 01 01 01 00 16 05 04 01 01 01 04 16 05 04 1B 2C EA 6B 3B 80 BA 54 D3 9D 70 33 14
PICC Free Memory: 1760
PICC Change Key ID: 00
PICC Config Changeable: true
PICC Free Create Delete: true
PICC Free Directory List: true
PICC Key Changeable: true
PICC Max Keys: 01
PICC Key 0 Version: A2
Application Count: 2
Application IDs: 01 00 00 02 00 00
```

This causes the "Can not parse file" screen, and nothing can be done with this file.
So I suggest to accept such file (as it was written by the firmware previously) and only emulate the UID, no apps.
I will provide a simple PR for that.

### Reproduction

1. Switch on
2. Go to NFC, Read
3. Place a specific Mifare DESFire token under the device
4. Save the data to a `.nfc`file
5. Go to NFC, Saved
6. Select the file that was just written
7. "Can not parse file" is shown, nothing can be done with the file.

### Anything else?

There is a follow-up bug - when trying to emulate the UID of that specific card immediately after the "Read", the external reader will _not_ accept the emulation; and this apparently continues until the Flipper is rebooted, also when using the data from the saved file (which I did after applying the mentioned PR).

LowSkillDeveloper, 2022-11-22:
I also have this problem with my bus pass, wrote here: https://github.com/flipperdevices/flipperzero-firmware/issues/1904#issuecomment-1295952022

skotopes, 2022-11-28:
Fixed in #2041 thanks to @mh-

PR #2041 (2022-11-28): NFC: Accept non-parsed apps in Mifare DESFire.
Fixes #2040

# What's new

Accept non-parsed apps in Mifare DESFire.

# Verification 

Load this file from "Saved" in the NFC app
```
Filetype: Flipper NFC device
Version: 2
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic
Device type: Mifare DESFire
# UID, ATQA and SAK are common for all formats
UID: 04 1B 2C EA 6B 3B 80
ATQA: 44 03
SAK: 20
# Mifare DESFire specific data
PICC Version: 04 01 01 01 00 16 05 04 01 01 01 04 16 05 04 1B 2C EA 6B 3B 80 BA 54 D3 9D 70 33 14
PICC Free Memory: 1760
PICC Change Key ID: 00
PICC Config Changea [... cut]
commit 1b315652 (2022-11-28): NFC: Accept non-parsed apps in Mifare DESFire. (#2041)

* NFC: Accept non-parsed apps in Mifare DESFire. Fixes #2040
closed 2022-11-28 by skotopes