flipperdevices/flipperzero-firmware #2046: Crash in iButton emulation

Tobirg, 2022-11-24:
### Describe the bug.

I found a bug on the Z5R, I also tried it on the Metakom intercom, 
but it is much more difficult to catch it there.
In the Telegram chat, people also reported this problem with other intercoms.
I hope the [video](https://youtu.be/DQ743CDfes4) will explain the problem

### Reproduction

If you run key emulation with the correct CRC and touch the reader, then the bug will be harder to catch.
But if you hold Flipper on the reader **before** starting the emulation, the bug will appear with almost 100% probability.

If you use any key with the wrong CRC, the bug can be caught in all cases: 
launch -> touch,
launch -> hold,
hold -> launch.

Tobirg, 2022-11-27:
Another one
https://t.me/flipperzero/103924/123382

DrZlo13, 2022-12-01:
Fixed https://github.com/flipperdevices/flipperzero-firmware/pull/2074

closed 2022-12-01 by DrZlo13