flipperdevices/flipperzero-firmware #2057: FA not picking up certain frequencies that the read raw can.

DarkDestroyerIII, 2022-11-27:
Tested with gs75 Lexus car key, fccid Hyq14fba. The flipper's FA is unable to detect the signal sent from the car clicker, however, the read raw function of the flipper is. The flipper's fa was unable to also detect signals sent from a 2020 ford, however, I do not know the exact make of the car.  I tested this with base firmware, and even dfu'd my flipper to test.

My discord is Norvarg#0425,  you can ask me if you need additional info.

maximed911, 2022-11-28:
i have the same problem.. some car work and some don't...

skotopes, 2022-12-01:
FA got some limitations: signal should be long enough and stay at the same frequency. Some fobs emit too short signal or using FSK with big deviation that makes them hard to be detected. In that case check FCC-ID on the back,  google it and check frequency in certification reports.

In general we are intentionally ignoring cars protocols to prevent device misuse. Also most of modern ones are rolling codes which makes copying them meaningless.

closed 2022-12-01 by skotopes