flipperdevices/flipperzero-firmware #2059: General imrovements from Model-7

Supertera, 2022-11-27:
### Description of the feature you're suggesting.

[image]
I think the developers of the flipper are familiar with a device called "model-7". This device has a lot of manufacturing codes and protocols that are not implemented in the flipper. Question - why? There is no way to contact one of the developers of model-7? To be honest, I thought that the developers of the flipper and the model-7 were the same people. I'm wrong? Tell me, why reinvent the wheel and add one protocol and a manufacturing code to the flipper firmware release, when you could combine the functionality with model-7 and make a truly amazing tool! My advice is to find the right person, especially with such funding...

skotopes, 2022-11-27:
Nope, we got nothing to do with it.

Supertera, 2022-11-27:
Well, yes. And never held third-party code grabbers in your hands :-)

For anyone wondering, this is the list of Sub-Ghz protocols the Codegrabber Model 7 can understand:-

Allmatic B.RO OVER
Allmatic B.RO STAR
Allmatic B.RO WN
Allmatic TECH 3
AN-MOTORS Black and Silver Buttons
APRIMATIC TR2 / TR4
BENINCA T4WV
BENINCA CUPIDO 4
BENINCA TO GO 2 / 4WV
BYTEC / BAISHENG
Bytec TB-4
Baisheng RC4-T15
BFT MITTO 2/4, 2N / 4N – Two modes of operation with these BFT MITTO consoles:
– Fast (regular copying), may not work in 100% of cases
– Universal mode – with data collection and calculation on a PC.
CAME TAM 432
CAME TOP 432/434
CAME TOP 432/434 EE
CAME ATOMO AT 01/02/04
CAME TOP 862NA (868Mhz)
COMUNELLO KEEP 2/4
COMUNELLO KEEP VICTOR 2/4
Doorhan Transmitter 4/2 / D2
Doorhan RCBLACK-4
DTM NEO
DEA Mio tR 2/4
DEA 273 (grey buttons, grey buttons)
DITEC GOL4
DITEC GOL4C (red LED, red LED)
DITEC BIXLS2
FAAC XT4 433 RC
FAAC TM1 / 2/3-DS
FAAC 868SLH only operates at 868.35 MHz. (via code substitution)
FAAC XT2 / XT4 868SLH – Requires calculation of the code on the PC
Marantec D222 / D224 / 214 (868Mhz)
Marantec Digital 302/313/304 (868Mhz)
NORMSTAHL / CRAWFORD RCU433-4K, FHS10-01 / 02, EA433-4KS / 2KS, EA433-2KM, N002800, T40-1 / 1M, T40-4 / 4M
MOOVO MT4 / MT4G / MT4V
Nice Flo 4/2/1
Nice Flor-s (-m) 4/2/1
Nice Very VE
Nice Very VR
Nice Smilo 2/4, blue/black
Nice era flor
Nice Era Inti (INTI1x / INTI2x)
Nice Era One (ON1E … ON9E) fixed the problem with opening
NERO Radio 8101-½ / 4 (434Mhz)
NERO Radio 8101-2М (434Mhz)
NOVOFERM Novotron 302/502
ROGER H80 TX22
ROGER H80 TX12 / 14
GENIUS BRAVO
GENIUS ECHO TX4
GSN TXRC09
GSN TXRC10
GSN TxRC09TB
GSN TxRC10TB
GSN TR-300S
GSN TR-500F
ELMES UMB
ELMES DWB and other compatible
HORMANN HSM4, HSE2, HSP4 (868Mhz) (Blue buttons)
Keeloq systems and many, many others …
Support for writing EM4305 / T5577 discs for encoding EM4100, T5577 for HID PROX II,

closed 2022-11-27 by skotopes