flipperdevices/flipperzero-firmware #2060: FBT' deps can't be built with Python 3.11

boozook, 2022-11-27:
### Describe the bug.

Python 3.11 now default on Github CI.
Some dependencies of FBT (see requirements.txt) throws error by setup.py because it doesn't migrated to Python 3.11.

For [example](https://github.com/boozook/flipper0/actions/runs/3547799386/jobs/5958330765#step:9:39) there's `heatshrink2`.

### Reproduction

1. update Python or just use default Python in Github CI
2. checkout repo
3. install dependencies following [documentation][], run `pip3 install -r scripts/requirements.txt`

[documentation]: https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/fbt.md#requirements

### Target

native, host

### Logs

```Text
2022-11-25T12:12:50.6016600Z Collecting heatshrink2
2022-11-25T12:12:50.6474550Z   Downloading heatshrink2-0.11.0.tar.gz (110 kB)
2022-11-25T12:12:50.6577670Z      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 110.6/110.6 kB 10.4 MB/s eta 0:00:00
2022-11-25T12:12:50.6978610Z   Preparing metadata (setup.py): started
2022-11-25T12:12:51.4558000Z   Preparing metadata (setup.py): finished with status 'done'
2022-11-25T12:12:51.5159250Z Installing collected packages: heatshrink2
2022-11-25T12:12:51.5261290Z   DEPRECATION: heatshrink2 is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
2022-11-25T12:12:51.5262180Z   Running setup.py install for heatshrink2: started
2022-11-25T12:12:59.8148940Z   Running setup.py install for heatshrink2: finished with status 'error'
2022-11-25T12:12:59.8351090Z   error: subprocess-exited-with-error
2022-11-25T12:12:59.8452920Z   
2022-11-25T12:12:59.8554820Z   × Running setup.py install for heatshrink2 did not run successfully.
2022-11-25T12:12:59.8657720Z   │ exit code: 1
2022-11-25T12:12:59.8759640Z   ╰─> [103 lines of output]
2022-11-25T12:12:59.8862160Z       running install
2022-11-25T12:12:59.8964400Z       /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
2022-11-25T12:12:59.9066850Z         warnings.warn(
2022-11-25T12:12:59.9168310Z       running build
2022-11-25T12:12:59.9278160Z       running build_py
2022-11-25T12:12:59.9379610Z       creating build
2022-11-25T12:12:59.9489060Z       creating build/lib.macosx-10.9-universal2-cpython-311
2022-11-25T12:12:59.9591190Z       creating build/lib.macosx-10.9-universal2-cpython-311/heatshrink2
2022-11-25T12:12:59.9591980Z       copying heatshrink2/streams.py -> build/lib.macosx-10.9-universal2-cpython-311/heatshrink2
2022-11-25T12:12:59.9694400Z       copying heatshrink2/version.py -> build/lib.macosx-10.9-universal2-cpython-311/heatshrink2
2022-11-25T12:12:59.9797070Z       copying heatshrink2/__init__.py -> build/lib.macosx-10.9-universal2-cpython-311/heatshrink2
2022-11-25T12:12:59.9898850Z       copying heatshrink2/__main__.py -> build/lib.macosx-10.9-universal2-cpython-311/heatshrink2
2022-11-25T12:13:00.0000220Z       running build_ext
2022-11-25T12:13:00.0101780Z       building 'heatshrink2.core' extension
2022-11-25T12:13:00.0203800Z       creating build/temp.macosx-10.9-universal2-cpython-311
2022-11-25T12:13:00.0305990Z       creating build/temp.macosx-10.9-universal2-cpython-311/heatshrink2
2022-11-25T12:13:00.0408030Z       creating build/temp.macosx-10.9-universal2-cpython-311/heatshrink2/_heatshrink
2022-11-25T12:13:00.0510480Z       clang -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch arm64 -arch x86_64 -g -Iheatshrink2/_heatshrink -I/Library/Frameworks/Python.framework/Versions/3.11/include/python3.11 -c heatshrink2/_heatshrink/heatshrink_decoder.c -o build/temp.macosx-10.9-universal2-cpython-311/heatshrink2/_heatshrink/heatshrink_decoder.o
2022-11-25T12:13:00.0613270Z       clang -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch arm64 -arch x86_64 -g -Iheatshrink2/_heatshrink -I/Library/Frameworks/Python.framework/Versions/3.11/include/python3.11 -c heatshrink2/_heatshrink/heatshrink_encoder.c -o build/temp.macosx-10.9-universal2-cpython-311/heatshrink2/_heatshrink/heatshrink_encoder.o
2022-11-25T12:13:00.0715530Z       heatshrink2/_heatshrink/heatshrink_encoder.c:394:11: warning: code will never be executed [-Wunreachable-code]
2022-11-25T12:13:00.0816780Z           (void)hse;
2022-11-25T12:13:00.0918100Z                 ^~~
2022-11-25T12:13:00.1019970Z       heatshrink2/_heatshrink/heatshrink_encoder.c:399:11: warning: code will never be executed [-Wunreachable-code]
2022-11-25T12:13:00.1121560Z           (void)hse;
2022-11-25T12:13:00.1223330Z                 ^~~
...
```

GMMan, 2022-11-28:
Have you tried fbt's prepackaged toolchain instead of setting up your own?

boozook, 2022-11-28:
@GMMan,

> Have you tried fbt's prepackaged toolchain instead of setting up your own?

I'm not sure what toolchain do you mean. If that is ARM toolchain, so it's absolutely doesn't matter because the issue about steps before any ARM-toolchain usage. See steps to reproduce described above.
- checkout repo
- `pip3 install -r scripts/requirements.txt` _- ⬅ we are here_
- __Error__
- `./fbt` (it will download ARM-toolchain, then build fw) - ⬅ here is toolchain

Probably I'm not know something, so didn't understand you right 🤷‍♂️.

GMMan, 2022-11-28:
fbt will download everything needed to build on its own, including its own copy of Python and the ARM toolchain. If you're only trying to set up an environment for fbt, it is not necessary.

boozook, 2022-11-28:
>  including its own copy of Python

Wow! So then I should just skip `pip install -r requirements.txt`?
I'll try, thanks.

boozook, 2022-11-28:
So I'm closing this issue, but someone should update deps to support Python 3.11 anyway.

closed 2022-11-28 by boozook