flipperdevices/flipperzero-firmware #2061: Weather App Temp in Fahrenheit

Crsarmv7l, 2022-11-28:
### Describe the enhancement you're suggesting.

Certain weather sensors (like inFactory) natively transmit in degrees F. inFactory in the weather app always show degrees C.

It would be nice be able to choose if you want F or C.

Skorpionm, 2022-12-05:
I'm looking forward to per-application customization support functionality. then I’ll add this so that I don’t have to re-select the settings every time I start the application

Crsarmv7l, 2022-12-05:
@Skorpionm 
While I could see how that would make things easier, I wasn't even thinking of actual persistent "settings", simply that right/left or up/down while viewing a weather station would cycle between F or C.

skotopes, 2022-12-20:
Fixed in dev, will be included in next release.

closed 2022-12-20 by skotopes