flipperdevices/flipperzero-firmware #2067: NFC: ATQA on the screen is not match ATQA in dump file

Mixaill, 2022-11-30:
### Describe the bug.

ATQA on the flipper screen mismatch with the ATQA inside the dump file (bytes are swapped)
@gornekich 

1) flipper screen
[image]

2) dump file
```
...
# UID, ATQA and SAK are common for all formats
UID: E9 1C F5 3C
ATQA: 04 00
SAK: 08
# Mifare Classic specific data
Mifare Classic type: 1K
Data format version: 2
# Mifare Classic blocks, '??' means unknown data
Block 0: E9 1C F5 3C 3C 08 04 00 62 63 64 65 66 67 68 69
```

### Reproduction

1. read NFC tag
2. save it to the file
3. open dump info on the flipper zero screen
4. open dump file on the PC
5. compare ATQA

### Target

0.72.0-rc

gornekich, 2022-12-01:
Hello @Mixaill 
Thanks for bug report. Will fix it soon

Astrrra, 2022-12-08:
Fixed in #2098, please create a new issue if the problem persists.

closed 2022-12-08 by Astrrra