flipperdevices/flipperzero-firmware #2068: NFC/Magic: magic write application does not check UID check byte

Mixaill, 2022-11-30:
### Describe the bug.

NFC Magic application does not check that dump contains valid UID check byte which allows to make magic tag unreadable.

### Reproduction

1. dump empty magic tag
2. download it on PC
3. change UID without changing check byte
4. upload file back to flipper
5. flash magic card
6. card becomes non-readable

doomwastaken, 2023-11-03:
Was unable to reproduce, reopen if issue persists

closed 2023-11-03 by doomwastaken

prooletsearch, 2024-02-11:
I find the app incredibly useful! Thanks for investing your time in creating it! I have saved one bricked magic card, not even with ICopy-XS device I was able to do that.