flipperdevices/flipperzero-firmware #2069: NFC: add ability to read cards with invalid check byte

Mixaill, 2022-11-30:
### Describe the enhancement you're suggesting.

See https://github.com/flipperdevices/flipperzero-firmware/issues/2068

After changing NFC UID to the invalid value flipper is not able to read the tag. 
It would be nice to have possible to bring tag back to life.

### Anything else?

Repro:
1. Brick card as described to https://github.com/flipperdevices/flipperzero-firmware/issues/2068
2. Try to read tag via flipper
3. Flipper does not detect the tag in NFC menu
4. However, magic application can detect it, but write operations does not detect the tag

skotopes, 2022-12-01:
@gornekich @Astrrra also make sense to bring this topic to hackathon

Mixaill, 2023-04-02:
Partially fixed in https://github.com/flipperdevices/flipperzero-firmware/pull/2511

bbk79, 2023-08-09:
I have a similar issue with Mifare UL magic tags, not sure which generation but they're not detected as supported tags by Magic NFC. I managed to successfully change the UID on them using NFC CLI APDU direct writes and also managed to brick a few in the process before figuring BCC and how to write the first few blocks with a single command.

Tried modifying the NFC CLI similar to [#2511](https://github.com/flipperdevices/flipperzero-firmware/pull/2511) and disable BCC checks w/o much luck, bricked tags are not readable nor detected at all in Magic NFC.