flipperdevices/flipperzero-firmware #2071: Ambient Weather TX-8300 to Weather Station

sb-cl, 2022-12-01:
### Description of the feature you're suggesting.

Add support for Ambient Weather TX-8300 (TFA 30.3211.02) sensor as it is currently not supported.

Reference implementation is available in `rtl_433` [project repository.](https://github.com/merbanan/rtl_433/blob/5f0ff6db624270a4598958ab9dd79bb385ced3ef/src/devices/ambientweather_tx8300.c)

skotopes, 2022-12-01:
Can you provide raw recordings of your sensor?

sb-cl, 2022-12-01:
Channel: 1 ; Battery: 0 ; Temperature: 25.6 C ; Humidity: 33 %
[g001_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10130062/g001_433.92M_250k.cu8.zip)

Channel: 1 ; Battery: 2 ; Temperature: 9.8 C ; Humidity: 32 %
[g003_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10130066/g003_433.92M_250k.cu8.zip)

Channel: 2 ; Battery: 0 ; Temperature: 7.2 C  ; Humidity: 35 %
[g002_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10130076/g002_433.92M_250k.cu8.zip)

Is this sufficient?

skotopes, 2022-12-01:
Yep, that should be enough for a start. @Skorpionm will take a look on it.

sb-cl, 2022-12-05:
Not sure if those were recorded correctly.
[subghz.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10151072/subghz.zip)
Readings should be like:
Channel: 2 ; Temp: -11.9 C ; Humidity: 53 %

Skorpionm, 2022-12-12:
spit twisted on every your data. the protocol is similar to https://github.com/merbanan/rtl_433/blob/5f0ff6db624270a4598958ab9dd79bb385ced3ef/src/devices/ambientweather_tx8300.c the format approximately converges, but RTL433 does not perceive it. and the data encoded in your package does not compete with the data presented. you need more records (.C8 is suitable) but with negative and positive temperatures, different humidity and other things, at least 10 pieces and accurate receiver readings (do not confuse)

and if possible, check if it shows RTL433, maybe I'm doing something wrong. it is very embarrassing that there is only 1 data sending in the record, usually 4+ repetitions go

sb-cl, 2022-12-13:
Command line `rtl_433 -S known -R 112`

```
model     : AmbientWeather-TX8300                  id        : 12
channel   : 2            Battery   : 2             Temperature: 17.9 C       Humidity  : 62 %
MIC       : CHECKSUM
```
[g001_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10215039/g001_433.92M_250k.cu8.zip)

```
model     : AmbientWeather-TX8300                  id        : 12
channel   : 2            Battery   : 2             Temperature: 19.7 C       Humidity  : 86 %
MIC       : CHECKSUM
```
[g002_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10215058/g002_433.92M_250k.cu8.zip)

```
model     : AmbientWeather-TX8300                  id        : 12
channel   : 2            Battery   : 2             Temperature: 19.6 C       Humidity  : 95 %
MIC       : CHECKSUM
```
[g003_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10215072/g003_433.92M_250k.cu8.zip)

```
model     : AmbientWeather-TX8300                  id        : 12
channel   : 2            Battery   : 2             Temperature: 22.3 C       Humidity  : 97 %
MIC       : CHECKSUM
```
[g004_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10215110/g004_433.92M_250k.cu8.zip)

```
model     : AmbientWeather-TX8300                  id        : 56
channel   : 1            Battery   : 2             Temperature: 8.4 C        Humidity  : 92 %
MIC       : CHECKSUM
```
[g005_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10215218/g005_433.92M_250k.cu8.zip)

```
model     : AmbientWeather-TX8300                  id        : 56
channel   : 1            Battery   : 2             Temperature: -6.6 C       Humidity  : 58 %
MIC       : CHECKSUM
```
[g006_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10215554/g006_433.92M_250k.cu8.zip)

```
model     : AmbientWeather-TX8300                  id        : 32
channel   : 3            Battery   : 0             Temperature: -3.0 C       Humidity  : 69 %
MIC       : CHECKSUM
```
[g007_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10215559/g007_433.92M_250k.cu8.zip)

```
model     : AmbientWeather-TX8300                  id        : 32
channel   : 3            Battery   : 0             Temperature: 1.4 C        Humidity  : 83 %
MIC       : CHECKSUM
```
[g008_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10215564/g008_433.92M_250k.cu8.zip)

```
model     : AmbientWeather-TX8300                  id        : 88
channel   : 3            Battery   : 0             Temperature: 6.2 C        Humidity  : 89 %
MIC       : CHECKSUM
```
[g009_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10215579/g009_433.92M_250k.cu8.zip)

```
model     : AmbientWeather-TX8300                  id        : 88
channel   : 3            Battery   : 0             Temperature: 11.6 C       Humidity  : 96 %
MIC       : CHECKSUM
```
[g010_433.92M_250k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10215600/g010_433.92M_250k.cu8.zip)

Skorpionm, 2022-12-13:
good!

Skorpionm, 2022-12-13:
understood what was the problem, you recorded data at a speed of 250k per second. URH won't let me reproduce data slower than 2M you can draw another rapka at a speed of 2M

Skorpionm, 2022-12-13:
Thanks. nice to work with you. check)
https://github.com/flipperdevices/flipperzero-firmware/pull/2116

sb-cl, 2022-12-15:
Thank you, seems to be working. Once caught a somewhat like assert, will try to get a reproducible sequence of actions.

skotopes, 2022-12-15:
Merged in dev, will be included in next release.

closed 2022-12-15 by skotopes