flipperdevices/flipperzero-firmware #2072: ST25TB nfc tag support

augustozanellato, 2022-12-01:
### Description of the feature you're suggesting.

It would be nice to have support for ST25TB nfc tags. They're supported by [ST's RFAL](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/lib/ST25RFAL002/include/rfal_st25tb.h) so it shouldn't be hard to add support at least in `furi_hal_nfc`. 
I'd need the support for a FAP I'm developing that reads, parses and writes back data to various types of tags so I don't need support in flipper gui etc.
I wouldn't mind using RFAL directly myself but if I understood everything correctly it isn't supported nor recommended.

skotopes, 2022-12-01:
Yep, RFAL is obsolete and going to be removed. @gornekich can provide more details...

gornekich, 2022-12-01:
Hello @augustozanellato !
As @skotopes said, it's not recommended to use RFAL. I am working on new low level nfc driver. Once it's done (hopefully end of december 2022) I will let you know

augustozanellato, 2022-12-01:
> Hello @augustozanellato ! As @skotopes said, it's not recommended to use RFAL. I am working on new low level nfc driver. Once it's done (hopefully end of december 2022) I will let you know

Thanks for the quick reply!

augustozanellato, 2023-11-01:
Support added with #3076

PR #3076 (2023-10-16): Add ST25TB poller
Closes #2072

# What's new

- ST25TB poller
- ST25TB support in NFC app (only reading and saving)

# Verification 

- Scan st25tb tags in various sizes/versions
- Save them
- Load them back up

# Checklist (For Reviewer)

- [ ] PR has description of feature/bug or link to Confluence/Jira task
- [ ] Description contains actions to verify feature/bugfix
- [ ] I've built this code, uploaded it to the device and verified feature/bugfix
closed 2023-11-01 by augustozanellato