flipperdevices/flipperzero-firmware #2083: Subghz fails with out error with a zero in .sub file

evilpete, 2022-12-03:
### Describe the bug.

If a '0' is contained in the dot-sub tile's RAW_Data transmission will fail without error.

eg:
>RAW_Data: **0** -48 60 -44 60 -48 52 -52 56 -48 52 -52 52 -52 52 -52 52 -52 52 -52 52 -52

Flipper will not reported error and display transmission animation as normal

FSK will transmit an unmodulated burst, OOK does not appear to transmit anything

No Error is alert is displayed 

### Reproduction

download attached files
rename to from `Ford_TPMS-error.sub.txt` to  `Ford_TPMS-error.sub.txt`
and `Ford_TPMS-good.sub.txt` to `Ford_TPMS-good.sub.txt`

and load files  onto Flipper

run SDR software like rlt_sdr to look at the transmission 

this is transmitted with the `Ford_TPMS-error.sub` that contains a '0' in the data:

[image]

this is transmitted with the  file `Ford_TPMS-good.sub` without  '0' in the data:

[image]

skotopes, 2022-12-03:
0 is invalid value that never should appear in the stream. How did you got it there?

evilpete, 2022-12-03:
> 0 is invalid value that never should appear in the stream. How did you got it there?

wrote a script to convert `.ook` files generated by rtl_433 ([subghz_ook_to_sub.py](https://github.com/evilpete/flipper_toolbox/blob/main/subghz_ook_to_sub.py)) to `.sub` file and stumbled onto the bug.

(yes I fixed my code,  but figured the lack of error report with subghz was also a issue)

skotopes, 2022-12-04:
@Skorpionm can we add extra checks?

Skorpionm, 2022-12-05:
we can throw out unwanted values ​​and continue playing the file, but this simply cannot happen during normal recording of a RAW file. and if people write scripts to generate a RAW file, it would be logical to think what the physical meaning of the duration is 0, and how could this even happen

evilpete, 2022-12-05:
@Skorpionm   quite a few people are writing scripts to generate a RAW file,  I have a repo half full of them

(Also it would have to have a official format description)

skotopes, 2022-12-10:
We'll add documentation for RAW format in next release.

closed 2022-12-10 by skotopes