flipperdevices/flipperzero-firmware #2085: NFC descripter with computer

colinborins, 2022-12-03:
### Description of the feature you're suggesting.

Create a way to help decrypt NFC via a flipper with a computer. For example, there is a cloner software that currently does this with their product when they can not decrypt against. The program is called Icopy.

skotopes, 2022-12-10:
What exactly do you want to decrypt?

colinborins, 2022-12-15:
It would decrypt the  nfc tag using the computer and uses the flipper as a nfc reader for live decripting. I’ve found that the flipper always has problems decoding my mifare classic 1k and maybe using a computer attachement might be a helpful workaround for decoding the keys

skotopes, 2022-12-15:
What you want sounds like mfkey32 attack which already implemented and can be done with mobile or web companions.

closed 2022-12-15 by skotopes

colinborins, 2022-12-17:
but its not. Im talking about with the key its self. That methoid requires the reader. What i am suggesting would be like this video https://www.youtube.com/watch?v=0SewclaC8Y4 but with the flipper instead.

skotopes, 2022-12-17:
Everything that is shown in this video is already implemented in Rfid, Nfc and Nfc magic apps.