flipperdevices/flipperzero-firmware #2094: buffered_file_stream_open never returns when trying to open executable itself

QtRoS, 2022-12-05:
### Describe the bug.

Facing this issue in my Hex Viewer app. When trying to view hex_viewer.fap itself, app hangs. Debug showed that `buffered_file_stream_open` (internally `file_stream_open`) function doesn't return control. Code is [here](https://github.com/QtRoS/flipper-zero-hex-viewer/blob/master/hex_viewer.c#L161). Seems that file is blocked by OS/app launcher. 

### Reproduction

1. Open [Hex Viewer app](https://github.com/QtRoS/flipper-zero-hex-viewer/) 
2. Navigate to the apps folder
3. Select hex_viewer.fap
4. App hangs, doesn't react to any input (while OS itself is alive, logs are going, qFlipper works)

skotopes, 2022-12-05:
That's not a bug, that is a feature. Opening already open file is currently impossible.

DrZlo13, 2022-12-05:
@skotopes yes, but in the case of an elf file, we don't need to keep that file open.