flipperdevices/flipperzero-firmware #2109: Subghz - Roll-over raw read

LoZio, 2022-12-09:
### Describe the enhancement you're suggesting.

Need: record a signal you do not know when will happen, but you know freqency and modulation
Current status: you need to keep the recording function running until asome limit is reached. The you get a big file that is 99.999% blank.
Suggestion: set a roll-over time, say 5 seconds, and FZ keeps recording and discarding the recording if no signal (rss<) is found. If the signal is recorded during that time a file is saved and recording stops, or optionally it loops saving recording-001,002,...

This can be obtained tweaking current application. The config only needs 4 more switches:
Recording type: continuous (current behavior) | loop
Loop duration: 1,5,10,.. seconds
Action on signal recorded: stop | loop to another file
RSS: squelch level

skotopes, 2022-12-10:
Hi. We already have RAW for that, but we may add something like that in future. Meanwhile I'll add this issue to backlog.

Skorpionm, 2022-12-10:
something similar has already been implemented.  run read raw , config, set threshold rssi, start recording.  only signals above the threshold rssi will be recorded.  the rest will not be written.  You can stay in this mode for as long as you like.

LoZio, 2022-12-11:
Thank you @Skorpionm I will try this mode too even if the use case is not the same I have in mind.
This also stresses the need for a detailed documentation of FZ firmware function. I understand this is still a wildly moving taget but may also result in forum threads and issues here that can be avoided.
I hope I can help in future with documentation.