flipperdevices/flipperzero-firmware #2110: Main menu customization via settings

makkarpov, 2022-12-09:
### Description of the feature you're suggesting.

Every user has his own set of frequently used apps, including custom ones. Currently custom apps must be accessed either via BtnDown->'Favorites', or via 'Main menu' -> 'Applications' -> ... (too many clicks). And vise versa, you could have unused app in main menu.

Suggestion is to allow customization of main menu via 'Settings', and leave just one fixed entry - 'Settings' itself.

skotopes, 2022-12-10:
Hi. We may add it at some point in future, meanwhile I'll add it to backlog.

makkarpov, 2024-07-07:
But reassignment of the buttons wasn't a topic of this request. Request was about re-ordering and customization of the main menu itself.

skotopes, 2024-07-07:
@makkarpov reordering in main menu is quite problematic with its current architecture. There are no plans to change it (also our docs department very unhappy when we change something they've documented).