flipperdevices/flipperzero-firmware #2114: Add Oregon-v-1 and wrong Humidity at Nexus-TH

oldbug62, 2022-12-10:
### Description of the feature you're suggesting.

Is it possible to ad Oregeon-v-1 Sensor? It is availabel in the rtl_433 plugin. Thx

### Anything else?

Just an Info:

The Humidity at Nexus-TH is wrong calculated at the the Weather Plugin but also in the rtl_433 plugin.

skotopes, 2022-12-10:
Hi. Yes, but we need RAW recording of this sensor.
Same goes for incorrect Nexus-TH readings: provide us recordings with corresponding data.

oldbug62, 2022-12-10:
Here the files recorded with flipper, just delete the .zip
I have an rtl_sdr maybe its better to record it with that let me know what file format do you need.
[Oregon-v 1_RAW_20221210-102455.sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10199833/Oregon-v.1_RAW_20221210-102455.sub.zip)
[Nexus-TH_RAW_20221210-105426.sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10199834/Nexus-TH_RAW_20221210-105426.sub.zip)
[Nexus-TH_RAW_20221210-104839.sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10199835/Nexus-TH_RAW_20221210-104839.sub.zip)
[Oregon-v 1_RAW_20221210-102957.sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10199836/Oregon-v.1_RAW_20221210-102957.sub.zip)

Skorpionm, 2022-12-10:
what do you mean not correct?
give your records with a photo of what the real receiver shows, so that there is something to focus on

what is the point of anonymized data

skotopes, 2022-12-10:
What @Skorpionm is trying to say is that each file must be also labeled with real values (the one that base station shows you)

oldbug62, 2022-12-10:
The Humidity displays 21% in the Flipper Wheater Station app. My Weatherstation show about 52-55% Humidity.

[image]

skotopes, 2022-12-10:
@oldbug62 please match it with one of the records you've provided or provide new ones that are matched.
Otherwise we can not properly verify it.

oldbug62, 2022-12-10:
[image]
[RAW_20221210-143542.sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10200193/RAW_20221210-143542.sub.zip)

Skorpionm, 2022-12-10:
https://github.com/flipperdevices/flipperzero-firmware/pull/2116
[image]
[image]
[image]
If >95 =>95
If <20 =>20
otherwise completely identical to the receiver

about Oregon-v-1

 I need to write down from the flipper and with photographs what the receiver shows, 5 pieces at different temperatures and humidity

oldbug62, 2022-12-10:
Okay, I have expressed myself incorrectly. 
The weather station does not display humidity, but the sensor transmits the humidity to the weather app of the Flipper and this does not match other displays. Maybe the sensor has a defect. Sorry

attached 5 raw files with different temperatures and pictures.

My oregeon v1 only has an indoor and outdoor temperature no humidity.
[image]
[image]
[image]
[image]
[image]
[RAW_20221210-213631 31,3C.sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10200890/RAW_20221210-213631.31.3C.sub.zip)
[RAW_20221210-212819 29,0C.sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10200891/RAW_20221210-212819.29.0C.sub.zip)
[RAW_20221210-212426 24,8C.sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10200892/RAW_20221210-212426.24.8C.sub.zip)
[RAW_20221210-211757 22,3C.sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10200893/RAW_20221210-211757.22.3C.sub.zip)
[RAW_20221210-211521 20,5C.sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10200894/RAW_20221210-211521.20.5C.sub.zip)

Skorpionm, 2022-12-13:
https://github.com/flipperdevices/flipperzero-firmware/pull/2116
add Oregon-v1 protocol

you need to test with pulling out and inserting the battery so that the sensor ID changes. you need to catch the moment when the high bit of the ID is 1 (the ID is visible in the data window) and whether the temperature will be parsed correctly.

closed 2023-02-10 by skotopes