flipperdevices/flipperzero-firmware #2123: Support Radio sonde recieving

BrunoKlu, 2022-12-12:
### Description of the feature you're suggesting.

Add feature \ app to recieve and decode weather ballon radio sonde signals

skotopes, 2022-12-14:
In theory possible in real life not going to work. Biggest issue is a lack of good antenna and LNA in flipper.

closed 2022-12-14 by skotopes

BrunoKlu, 2022-12-15:
Ok,Do you mean that Flippers reciever would not be sensitive enough ?

Radio sonde protocols are quite strong modulations, but you are probably right !

Thanks

skotopes, 2022-12-15:
Flipper receiver itself is quite good. But it's not enough: you also need good antenna, LNA and band-pass filter to get it really right.

daskalovst, 2024-04-25:
Yes, I agree. If we're using a flipper to receive signals from many kilometers away, it's definitely not the proper device for the task.

There are numerous software programs available that can predict where a radiosonde will fall to the ground. The concept here is to utilize the flipper to receive and decode the signal during the final stage of the radio-sonde's recovery, when we are in close proximity to it. At this point, the signal is very strong, making it more feasible for the flipper to successfully decode it. I believe the flipper can effectively decode signals from distances of 200-300 meters in this scenario.

skotopes, 2024-04-26:
@daskalovst by the way external subghz module support has been available for some time. I think at this point everything to implement sonde support is ready.

However this is not topic of interest for us, so someone from community should try to implement it.

pfrederiksen, 2025-09-04:
I'm working on a app for this, stay tuned.