flipperdevices/flipperzero-firmware #2130: Add La Crosse TX29DTH-IT support for Weather Station

kaimi-, 2022-12-14:
### Description of the feature you're suggesting.

Add La Crosse TX29DTH-IT support for Weather Station. 
rtl_433 supports this sensor: https://github.com/merbanan/rtl_433/blob/bb1be7f186ac0fdb7dc5d77693847d96fb95281e/src/devices/lacrosse_tx35.c.

Some samples (`rtl_433_64bit_static.exe -R 75 -f 868300000 -s 2M -S known`):
```
model     : LaCrosse-TX35DTHIT                     id        : 6
Battery   : 1            NewBattery: 1             Temperature: 23.9 C       Humidity  : 29 %          Integrity : CRC
```
[g001_868.3M_2000k.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10225110/g001_868.3M_2000k.zip)

```
model     : LaCrosse-TX35DTHIT                     id        : 6
Battery   : 1            NewBattery: 1             Temperature: 19.6 C       Humidity  : 26 %          Integrity : CRC
```
[g002_868.3M_2000k.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10225127/g002_868.3M_2000k.zip)

```
model     : LaCrosse-TX35DTHIT                     id        : 6
Battery   : 1            NewBattery: 1             Temperature: -0.6 C       Humidity  : 32 %          Integrity : CRC
```
[g003_868.3M_2000k.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10225223/g003_868.3M_2000k.zip)

```
model     : LaCrosse-TX35DTHIT                     id        : 6
Battery   : 1            NewBattery: 1             Temperature: -3.8 C       Humidity  : 35 %          Integrity : CRC
```
[g004_868.3M_2000k.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10225268/g004_868.3M_2000k.zip)

```
model     : LaCrosse-TX35DTHIT                     id        : 6
Battery   : 1            NewBattery: 1             Temperature: 19.0 C       Humidity  : 41 %          Integrity : CRC
```
[g005_868.3M_2000k.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10225419/g005_868.3M_2000k.zip)

Skorpionm, 2022-12-15:
[image]

transmission in FM modulation Deviation 68KHz, it is not supported in the flipper out of the box. and the recordings play crooked, I need a real sensor

skotopes, 2022-12-15:
More like you need to create custom preset and add it to subghz configuration files

kaimi-, 2022-12-16:
@Skorpionm probably I don't understand something, but why are you showing 433.925 frequency? This sensor uses 868.3 frequency, which can be deduced from recording filename.

Skorpionm, 2022-12-16:
just played on 433.92

kaimi-, 2022-12-19:
@Skorpionm  Could you tell anything about those recordings, still crooked? Both should contain the same data.
```
model     : LaCrosse-TX35DTHIT                     id        : 6
Battery   : 1            NewBattery: 0             Temperature: -1.9 C       Humidity  : 68 %
Integrity : CRC
```
[g001_868.3M_2000k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10257479/g001_868.3M_2000k.cu8.zip)
[g002_868.3M_2000k.cu8.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10257480/g002_868.3M_2000k.cu8.zip)

closed 2023-03-13 by skotopes