flipperdevices/flipperzero-firmware #2132: event.input.sequence returns the same wrong value on remote control

vad7, 2022-12-14:


skotopes, 2022-12-14:
Yep, external events doesn't hame valid sequence number.
Can I see use your case of it?

vad7, 2022-12-14:
switch(event.input.key) {
	case InputKeyOk:
		if(event.input.type == InputTypeLong) {
			...func#1
			key_press_seq_ok = event.input.sequence; 
		} else if(event.input.type == InputTypeRelease && key_press_seq_ok != event.input.sequence) {
			// short press
			...func#2
		}

skotopes, 2022-12-14:
ViewPort and View are ensuring that input sequence is always complimentary. Using event.input.sequence here looks like overkill.
There are some really special edge where you may want to use it, but I don't see any in your code.

Anyway, I'll update code to add sequence number for rpc input events.

vad7, 2022-12-14:
I need to handle long keypress and single keypress.
how to do it?

skotopes, 2022-12-14:
```
if(event.input.type == InputTypeShort) {
    // Short
} else if(event.input.type == InputTypeLong) {
    // Long
}
```

Something like that is not enough?

vad7, 2022-12-14:
Thanks, it's ok.

skotopes, 2023-06-06:
Couple releases ago I've implemented proper sequence number for external events. So your original issue should be solved.

closed 2023-06-06 by skotopes