flipperdevices/flipperzero-firmware #2135: Auto lock doesn't work on passport screen

RobinTail, 2022-12-14:
### Describe the bug.

I have configured `Auto Lock Time` to 30 seconds.
However, locking the device doesn't happen after the specified time of inactivity,
in case I active the passport screen by pressing the `Right` button.

I made a video recording of that behavior:

https://youtu.be/_-QI5vqfNbM

### Reproduction

1. Unlock
2. Press `Center` button
3. choose Settings
4. choose Desktop
5. choose Auto Lock Time
6. set to `30s`
7. Press `Back` 3 times to return to main screen
8. Press `Right`
9. Wait at least 30 seconds
10. Press `Back` to make sure the device is not locked yet

### Target

I have no idea what this is

### Anything else?

Firmware version `0.73.1`
SD card installed
Databases installed

skotopes, 2022-12-14:
That is expected behavior. Lock is only possible if no application is running. Passport is an application.

closed 2022-12-14 by skotopes

RobinTail, 2022-12-14:
Well, okay. Thank you for the quick response and the explanation.
I didn't expect that such a UX pattern, since similar the setting on me smartphone and laptop doesn't behave like that.

skotopes, 2022-12-14:
They(full-fledged OS) got much more advanced ways of managing processes and threads. We are closer to early 200x mobile phones: balancing between complexity and resource shortage.