flipperdevices/flipperzero-firmware #2136: Smart Electricity Meters: Infrared SML / OBIS

0x53A, 2022-12-14:
### Description of the feature you're suggesting.

Smart Meters broadcast their current status every X seconds through an IR transmitter using SML (Smart Message Language) / OBIS (Object Identification System) codes.

It would be really cool if you could record a raw transmission and then interpret it as SML / OBIS.

Here is one link about SML / OBIS: https://esphome.io/components/sml.html

skotopes, 2022-12-19:
Good idea for the app. Not sure if we are going to implement it.

rklasen, 2023-10-30:
This was closed as "completed", but I can't find this anywhere in the docs. Is the app to read a compatible power meter available?

skotopes, 2023-10-30:
As not planned.

closed 2023-10-30 by skotopes