flipperdevices/flipperzero-firmware #2145: HID Bluetooth problem

DiCollars, 2022-12-17:
### Describe the bug.

After updating (release 0.73.1) I have to re-pair my device every time to make work 'bluetooth remote' plugin (PC windows 11).
Have no idea why It was broken after update, please fix it - cos it was one of my favorite feature.

### Reproduction

1.Unpair all devices
2.Open on flipper plugin 'bluetooth remote' -> 'KeyBoard'
3.Open on PC settings - bluetooth devices -> add new one -> connect -> everething does work
4.Exit plugin on flipper
5.Open again this plugin
6.No PC reaction on anu command

DrZlo13, 2022-12-18:
Known issue, we a working on it rn.

skotopes, 2022-12-20:
Fixed in dev, going to be included in next release

closed 2022-12-20 by skotopes