flipperdevices/flipperzero-firmware #2153: Mifare Ultralite Unlock with Reader function hangs

tatramaco, 2022-12-19:
### Describe the bug.

The Unlock with Reader function for MiFare Ultralite hangs and Flipper must be restarted

### Reproduction

1. Read Mifare Ultralite card with password protected pages
2. Select More and Unlock
3. Choose Unlock With Reader
4. ‘Touch the reader to get password…’ message displayed
5. Hitting back before presenting to the reader works.
6. Presenting Flipper to the reader causes red LED for unsuccessful auth and it then hangs and must be rebooted

skotopes, 2022-12-20:
@gornekich please take a look

nils1603, 2022-12-29:
Can confirm that this is not working and causes in a freeze. Users on reddit also confirmed this:

https://www.reddit.com/r/flipperzero/comments/zwuip8/mifare_ultralight_c_unlock_with_reader_not_working/

GMMan, 2022-12-29:
> Users on reddit also confirmed this:
> 
> https://www.reddit.com/r/flipperzero/comments/zwuip8/mifare_ultralight_c_unlock_with_reader_not_working/

That's a separate issue, Mifare Ultralight C doesn't use the same authentication protocol.

nils1603, 2022-12-29:
I see. I am trying it with a Mifare Ultralight 11 card/reader

mitchellmebane, 2023-01-07:
I had the same hanging issue while trying to copy my key card for an Assa Abloy lock.  Based on the design, I think it was the [VingCard Essence](https://www.assaabloyglobalsolutions.com/en/hospitality-solutions/electronic-locks-for-hotels/vingcard-essence-for-hotels).  Unfortunately, I didn't save the card scan, and I'm not at that hotel any more, so I can't provide any more details - all I recall is that the Flipper identified it as a Mifare Ultralite something.

Robotto, 2023-01-30:
Confirmed with MF Ultralight 11 card.

dontic, 2023-02-13:
Hey @skotopes @gornekich Having the same issue here. Need to reset the flipper every time. Mifare Ultralight 11 as well.
Tried to get some debug logs but not sure what you guys need. Let me know and I can grab those while I'm at the hotel.
Here's the issue in a GIF:
[image]

dontic, 2023-02-13:
Huh, interesting thing happening here. 
I just read and saved the hotel key card. Then:

- If I select it and choose "Unlock With Reader", it hangs.
- If I select it and choose "Unlock With Password" and then "Unlock With Reader" it works just fine!  It detects the reader and then asks me to read the card again and works like a charm.

Sadly I have no J-Link here to log at all.

firstel, 2023-02-28:
@gornekich this looks like enough info to resolve?

skotopes, 2023-03-13:
@firstel yep

panki27, 2023-04-24:
I ran into this today.

The Flipper doesn't hang, I can quit out with the Back button. But "Unlock with Reader" does not seem to work, it stays in the "reading" phase.

If I use "Detect Reader" directly from the NFC menu, it gathers 10 nonces.

doomwastaken, 2023-11-03:
This was fixed a while back, also cannot reproduce on latest release (NFC refactor). Reopen the issue if problem persists

closed 2023-11-03 by doomwastaken