flipperdevices/flipperzero-firmware #2164: Improve detect reader documentation

bettse, 2022-12-20:
### Describe the enhancement you're suggesting.

In the flipper discord's #nfc channel we get more than a few people who have a mistaken impression about the Detect Reader function.  In at least one case they pointed back to the docs (https://docs.flipperzero.one/nfc) as to why.  I wanted to suggest some improvements to prevent this.

On the graphic:

[image]

Change "Receive reader's keys", suggestions:
 * "Perform MFKey32 attack"
 * "Emulate Mifare Classic 1k for MFKey32"
 * "Recover add'l Mifare Classic keys"

For the text below
> Detect Reader — emulates an NFC card to receive authentication keys from logs sent by a reader.

Suggestions:

 * **Detect Reader** -- emulates a Mifare Classic 1k to collect data (nonces) used to calculate keys attempted by reader
 * **Detect Reader** -- emulates a Mifare Classic 1k to collect nonces to calculate add'l keys used by reader

skotopes, 2022-12-20:
Thanks, forwarded to our technical writer. We'll update docs.

skotopes, 2022-12-20:
Done. Please check updated page.

closed 2022-12-20 by bettse

skotopes, 2022-12-20:
U R Welcome )