flipperdevices/flipperzero-firmware #2174: Flipper Zero App resets favorites when synced

alex1001458, 2022-12-22:
### Describe the bug.

Every application or file in your favorites tab the flipper zero app doesn't recognize in it's archive menu (eg. any app inside a folder) will get removed from the favorites tab when you sync it with the flipper zero.

### Reproduction

1. "Pin" something through the quick menu (press down on screen saver) the Flipper app doesn't recognize, such as an IR remote you have in a folder.
2. Turn on Flipper app and let it synchronize (you may have to press "synchronize" yourself)
3. Watch your favorited item disappear.

skotopes, 2022-12-22:
Are we talking about mobile apps?

alex1001458, 2022-12-22:
> Are we talking about mobile apps?

No, sorry for lack of clarification, I'm talking about any program you have on your flipper zero

skotopes, 2022-12-22:
Then what do you mean by "synchronize"?

skotopes, 2022-12-26:
@alex1001458 still need more details from you ;-) Please provide details on how to reproduce issue with exact firmware/platform/app versions been used.

alex1001458, 2022-12-28:
Sorry for lack of response. I'm just going to use pictures to simplify.
1.My Favorites before opening the phone app
[image]

2. When opening the phone app it only recognizes one of them.
[image]

3. The flipper Syncs to what the phone recognizes, removing the other 2 apps.
[image]

I suspect it's due to the fact the other 2 favorites are within folders
For example:
[image]

skotopes, 2022-12-28:
We talking about Android? Which version of firmware and companion app you are using?

alex1001458, 2022-12-28:
I am on Android, latest version of the companion app and I am using the unlshd-022 firmware.

skotopes, 2022-12-28:
Do you have this issue with original firmware?

alex1001458, 2022-12-29:
Yes.

[image]
[image]