flipperdevices/flipperzero-firmware #2216: IR flipper-to-flipper file transfer

Geddeon, 2022-12-29:
### Description of the feature you're suggesting.

Long-long time ago mobile phones has IR ports with file transferring feature. Why not add it to flipper? AFAIK there is no possibility of sharing files between flippers except for switching MicroSD cards.

ankris812, 2023-01-08:
or even ble is  better for that

skotopes, 2023-01-08:
In theory possible, practically meaningless. If we talking about IrDA than is kinda complex to implement, if something more simple than any other transport supported by flipper will be better.

closed 2023-01-08 by skotopes

nymda, 2023-01-08:
i disagree with "practically meaningless". being able to send NFC scans, nonces, etc wirelessly between two flippers could be very useful in actual pentesting applications

skotopes, 2023-01-09:
@nymda my point is that there are better transports for this task: mobile apps supports sharing on the internet, and if you need local sharing than NFC or Sub-GHz should be used as transport.

Anyway, we already have sharing implemented in mobile apps. Anything else is a good idea for an app and you are free to implement it ;-)

ankris812, 2023-01-09:
> @nymda my point is that there are better transports for this task: mobile apps supports sharing on the internet, and if you need local sharing than NFC or Sub-GHz should be used as transport. [...]

or even ble, im working on doing that. ble is more reliable than ir. and about nfc, do you mean something like android beam?