flipperdevices/flipperzero-firmware #2218: Cannot read EM4305 cards

gloglas, 2022-12-29:
### Describe the bug.

I cannot read the EM4305 cards.
Flipper reads EM4100 cards just fine and they should be compatible.

Here is a basic reading from proxmark and exterior appereance.

[image]
[image]

### Reproduction

Press Ok from homescreen
Go to 125kHz RFID menu
Press Read and hold card in various distances and positions

### Target

0.74.2

Astrrra, 2022-12-29:
Are you sure that it's an EM4305? Your proxmark screenshot shows that it's an EM4x50, when it should be EM4x05

gloglas, 2022-12-29:
This it the datasheet that online shop provided me with: http://www.advanide.com/wp-content/uploads/products/rfid/EM4205-4305.pdf
The chip that I bought was advertised as EM4305.
Either way flipper is unable to read it.

gloglas, 2022-12-29:
This is the EM4x05 readout and it is a EM4305.
That one above was set to auto search mode.

[image]

Astrrra, 2022-12-30:
Ok, it looks like your EM4305 is not emulating an EM4100 card, but some other one (that's what the EM4x50 proxmark reading suggests). Can you write an EM4100 payload to it using your proxmark and try reading that with the Flipper?

gloglas, 2022-12-30:
The EM4x50 is just a random noise. Please forget that I ever posted that.
I didn't successfully write EM4100 payload to the card but I am not sure if I did that correctly. So if you tell me what things you want me to try exactly _(iceman repo)_ I would be delighted to help you.

skotopes, 2023-01-08:
Nothing specific from our side, just try standard programming routine for this chip and ensure that payload is correctly emulated with proxmark.

closed 2023-01-08 by skotopes