flipperdevices/flipperzero-firmware #2231: Apple magsafe emulation

bettse, 2022-12-31:
### Description of the feature you're suggesting.

Very low priority, but I was thinking it would be fun to have an NFC menu item for emulation an Apple magsafe device like a phone case or wallet.  Or maybe also a format so one could scan existing, save, emulate, etc.  Lots of options, very minor benefit.  The magsafe devices appear as a type 4 tag with an NDEF record, the CC file contains bytes that indicate the specific color, device type (wallet/case), etc.  They use an unusual anti-collision, in theory to prevent overlap with regular 14a NFC.  Proxmark has lots of the anti-col details (https://github.com/RfidResearchGroup/proxmark3/blob/97e394c58a649d5f74d101202a76214e5e70d596/include/protocols.h#L163), and iceman rfid discord has some example CC in the back history.

skotopes, 2023-01-08:
Sounds interesting. May be a slight challenge since we delegate anti-collision to st25

dzienisz, 2023-01-10:
Qi2 will be made of Apple Magsafe https://www.youtube.com/watch?v=m-wWln9DA3Q he explain it!

ezhevita, 2024-02-13:
I’ve implemented MagSafe emulation for proxmark: https://github.com/ezhevita/proxmark3/tree/magsafe

However iPhone can’t read the emulation and my guess is that magnets in the case enable the NFC field and thus emulation will be useless without magnetic activation

skylarrgrey, 2024-03-07:
> I’ve implemented MagSafe emulation for proxmark: https://github.com/ezhevita/proxmark3/tree/magsafe
> 
> However iPhone can’t read the emulation and my guess is that magnets in the case enable the NFC field and thus emulation will be useless without magnetic activation

Could you put some MagSafe magnets on the iPhone and then test emulation ?

[image]

ihrapsa, 2024-04-20:
> I’ve implemented MagSafe emulation for proxmark: https://github.com/ezhevita/proxmark3/tree/magsafe
> 
> However iPhone can’t read the emulation and my guess is that magnets in the case enable the NFC field and thus emulation will be useless without magnetic activation

Hey @ezhevita I played with your fork and managed to emulate some data, however the phone does not react even with magnetic activation. below are the trace and commands I used. What am I doing wrong? As far as I could understand only the last 4 bytes differ in the capability container (CC).

```hf 14a sim -t 13 --data 57030390```
the data string I got is from Iceman's Discord server (@tcprst's message from 3 years ago) that corresponds to a _blue leather wallet_:

```
Orange Leather Case 12 Pro Max (Legit):
001D20010000FF0406E104030000000506E105008082830F044C01059B

Black Leather Case 12 Mini (Legit):
001D20010000FF0406E104030000000506E105008082830F044C01019F

White Case:
001D20010000FF0406E104030000000506E105008082830F0453010396

Black Case:
001D20010000FF0406E104030000000506E105008082830F0453010198

Blue Leather Wallet (Legit):
001D20010000FF0406E104030000000506E105008082830F0457030390

Black Leather Wallet (Legit):
001D20010000FF0406E104030000000506E105008082830F0457030192
```

Provided that the pm3 is positioned properly and then activated with a magnet over it (a 3rd party MagSafe wallet for example), I get a trace. However no animation from the phone. Here is the trace:

```
[usb] pm3 --> hf 14a sim -t 13 --data 57030390
[=] Press pm3 button to abort simulation
[#] Emulator stopped. Trace length: 489 
[=] Done!
[usb] pm3 --> trace list -t 14a
[+] Recorded activity ( 489 bytes )
[=] start = start of start frame. end = end of frame. src = source of transfer.
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |7a(7)                                                                    |     | MAGSAFE WUPA
       2228 |       4596 | Tag |04  03                                                                   |     | 
      11762 |      16530 | Rdr |50  00  57  cd                                                           |  ok | HALT
      39076 |      40068 | Rdr |7d(7)                                                                    |     | MAGSAFE WUPA
      41304 |      43672 | Tag |04  03                                                                   |     | 
      50836 |      53300 | Rdr |93  20                                                                   |     | ANTICOLL
      54472 |      60296 | Tag |00  00  00  00  00                                                       |     | 
      67476 |      78004 | Rdr |93  70  00  00  00  00  00  9c  d9                                       |  ok | SELECT_UID
      79176 |      82760 | Tag |20  fc  70                                                               |  ok | 
      90466 |      95234 | Rdr |50  00  57  cd                                                           |  ok | HALT
     117340 |     118332 | Rdr |7a(7)                                                                    |     | MAGSAFE WUPA
     119568 |     121936 | Tag |04  03                                                                   |     | 
     129116 |     131580 | Rdr |93  20                                                                   |     | ANTICOLL
     132752 |     138576 | Tag |00  00  00  00  00                                                       |     | 
     145756 |     156284 | Rdr |93  70  00  00  00  00  00  9c  d9                                       |  ok | SELECT_UID
     157456 |     161040 | Tag |20  fc  70                                                               |  ok | 
     168146 |     172914 | Rdr |50  00  57  cd                                                           |  ok | HALT
     184192 |     185184 | Rdr |7d(7)                                                                    |     | MAGSAFE WUPA
     186420 |     188788 | Tag |04  03                                                                   |     | 
     195952 |     206480 | Rdr |93  70  00  00  00  00  00  9c  d9                                       |  ok | SELECT_UID
     207652 |     211236 | Tag |20  fc  70                                                               |  ok | 
     218902 |     223670 | Rdr |e0  80  31  73                                                           |  ok | RATS - FSDI=8, CID=0
     224842 |     234122 | Tag |06  77  77  71  02  80  be  6a                                           |  ok | 
     246730 |     264106 | Rdr |02  00  a4  04  00  07  d2  76  00  00  85  01  01  a6  09               |  ok | 
     265342 |     271166 | Tag |02  90  00  f1  09                                                       |     | 
     278474 |     290090 | Rdr |03  00  a4  00  0c  02  e1  03  d2  af                                   |  ok | 
     291326 |     297150 | Tag |03  90  00  2d  53                                                       |     | 
     304458 |     313770 | Rdr |02  00  b0  00  00  0f  8e  a6                                           |  ok | 
[... cut ...]
     355964 |     377916 | Tag |03  05  06  e1  05  00  80  82  83  0f  04  57  03  03  90  90  00  24   |     | 
            |            |     |36                                                                       |  ok | 
     385174 |     388726 | Rdr |c2  e0  b4                                                               |  ok | 
     390986 |     395722 | Tag |ca  e0  74  ce    
```

[simulation_57030390.trace.zip](https://github.com/flipperdevices/flipperzero-firmware/files/15047964/simulation_57030390.trace.zip)

ezhevita, 2024-04-26:
> What am I doing wrong?

@ihrapsa I have no idea, I couldn’t make it work as well. I hope someone can take it from here since I am kinda busy with getting a job 😅

A200K, 2026-01-01:
@lerik4362 Happy new year!

First of all, I don't own an apple wallet with nfc functionality, so I'm not able to test or verify any of my statements.
The principle is quite simple I think, your iPhone will will save the location data at the time it loses connection to the nfc sticker, in order for you to be able to find it in "Find my".

You need a proxmark3 and a genuine Apple wallet in order to sniff the data that is transmitted. There is no way around it. If the protocol is as simple as it is for Apple's (genuine) iPhone cases, all you then need is to buy a NTAG424 sticker and write the data to the capability container file you saw during sniffing. It's likely there is an unique wallet identifier that you have to identify within the transmitted data, which will be registered in the "Find my" ecosystem. Maybe you even need a second genuine wallet, so you can see the differences between two wallets, in order to correctly identify the unique identifier part.
Your thirdparty wallet likely comes with the necessary magnet in order to trigger the protocol initiation, so you don't have to worry about that.

There is no way to know what the protocol looks like without owning at least one genuine wallet. Once someone reverse engineers it and shares the info, it could be possible we can choose our own unique wallet identifiers, and do exactly as you described.
It's also possible this wallet functionality is encrypted and requires secret keys. Then, you could clone a genuine wallet at best, but you wouldn't be able to create a new one, in order to use it with "Find my".

Unless you find someone that is able to do so, it is easier and cheaper to just buy and use a genuine apple wallet lol. Does anyone else in here own one? ;)

bettse, 2026-01-01:
"Find my" uses BLE/Bluetooth and is unrelated to the NFC magsafe stuff

mikkozy, 2026-02-06:
@A200K I have a number of NFC tags. Could you upload the CC files in order to upload them to the tag for animation? And is it possible to write this file from the phone and how? I'm sorry if I wrote something wrong, I'm very poorly versed in this topic. Thank you in advance!