flipperdevices/flipperzero-firmware #2234: Issue with Extra Actions read ASK and Pet Chip

f3rr311, 2023-01-01:
### Describe the bug.

Following Documentation https://docs.flipperzero.one/rfid/animal-microchips I was attepting to scan one of my three pets RFID tags they have from the Vet.

I can not get it to show up when using Scan ASK or Using NFC, I can However get it to scan as a RAW RFID and this works on all the tags. The pets are all over 6 years old so I am assuming they are not using the higher freq tags.

I can provide the RAW data if needed to figure out what's happening, I am assuming that for it to read the RAW data its in the right freq. and it's just a minor bug in the Scan ASK.

I will help however I can to figure out if its a me issue or a FW issue.

### Reproduction

1. Switch on
2. Goto RFID
3. Extra actions
4. Read ASK
5. Scan pet 

Works in Raw RFID
1. Switch on 
2. Goto RFID
3. Extra Actions
4. Read RAW RFID Data

### Target

RFID

DrZlo13, 2023-01-04:
RAW RFID is like "record audio via microphone", you can record noise or silence and it will be still valid record.

This mode is designed and used only to record an unknown tag data and pass it to the developer who will write the code to implement the protocol. To record a useful signal, you must make sure that the tag is directly in front of the flipper and does not move during the recording process.

skotopes, 2023-01-08:
Yep, ensure that your pet's ID chis is LF RFID and working. 
And then reopen with proper data.

closed 2023-01-08 by skotopes