flipperdevices/flipperzero-firmware #2241: Extending compatible weather stations to Acurite Iris 5-in-1

fiveseven808, 2023-01-04:
### Description of the feature you're suggesting.

Requesting adding compatibility to the Acurite Iris 5-in-1 weather stations. I assume it should be a similar protocol to the existing Acurite devices, but with more sensors to read off of. 

Manual of weather station in question is located here https://www.acurite.com/media/manuals/06014M-instructions.pdf
 This would be an extension of the existing weather station app. 

### Anything else?

I field several of these weather stations and would be able to test and collect data, however I'm unsure how to begin.

Skorpionm, 2023-01-05:
there is little sense from management.  real records from sensors are needed.

fiveseven808, 2023-01-05:
I have real sensors, but little experience obtaining data. How can I begin?

skotopes, 2023-01-08:
> I have real sensors, but little experience obtaining data. How can I begin?

Record RAW file with subghz app. You can find manual for adding new fobs/protocols on the forum.

Crsarmv7l, 2023-02-16:
> > I have real sensors, but little experience obtaining data. How can I begin?
> 
> Record RAW file with subghz app. You can find manual for adding new fobs/protocols on the forum.

Just curious, but do they have to be actual captures? rtl_433 has a plethora of test files, and after converting them to .ook with rtl_433, they can be converted to raw subs with evilpete's tools. I have converted huge amounts of rtl_433 captures (and files I have generated) with his tools, and they all work fine

fiveseven808, 2023-02-16:
Since my original post I've bought a couple of SDRs, and realized I am able to read my weather station just fine with rtl_433. 

Is there any way to port the protocol over from it?

skotopes, 2023-02-16:
@fiveseven808 yes, also we still need some recordings