flipperdevices/flipperzero-firmware #2242: Add space key to keyboard for (re)naming items

EinsteinMaker, 2023-01-04:
### Describe the enhancement you're suggesting.

### Requesting that a space key be added to the keyboard that appears when naming an item.

The keyboard doesn't have a space
[image]

The mobile app can be used to rename items with certain symbols, including spaces
[image]

SleepyYui, 2023-01-04:
I would personally say that using spaces in such applications is rather unusable, since whilst programming you can't just use space keys in variables. In general, creating files and folders with spaces in my opinion is not my thing. But to be honest I could see people using it since you can do it in the app.

EinsteinMaker, 2023-01-06:
I had no idea. idk why I didn't figure it out because I knew that you could long press letters but that's just me.

closed 2023-01-06 by EinsteinMaker

fiveseven808, 2023-01-06:
> I use Unleashed, but I am pretty sure it has to be working with the stock firmware too. To write space just long-press the OK button on the "_". (BTW you can also long-press on letters to make them UPPERCASE). [image]

Holy cow, I did not realize that this was a thing! I'm commenting to reccomend that long-press actions be added and detailed in the flipper zero documentation.

TinyTechPrints, 2024-09-07:
> I use Unleashed, but I am pretty sure it has to be working with the stock firmware too. To write space just long-press the OK button on the "_". (BTW you can also long-press on letters to make them UPPERCASE). [image]

hi, what is the ''_''????

timby3, 2024-09-07:
It's the button left of the "save" button.