flipperdevices/flipperzero-firmware #2243: opton of owner info on lock screen

evilpete, 2023-01-04:
### Description of the feature you're suggesting.

Are useful option would be to have the ability to put a phone number or email address onthe  sleep or lock screen so there might be a snowball's chance in hell of a  possibly be returned  if lost..

meeDamian, 2023-01-08:
In the meantime, consider writing contact info on the back/side and covering it with a bit of a transparent tape. And/or creating `IF_DEVICE_FOUND.txt` file on the microSD card.

bfgelectronics, 2023-02-01:
By any chance is someone trying to work on this feature and is it desired? If so, i would like to implement it (when my flipper arrives).

skotopes, 2023-02-10:
@bfgelectronics this topic was brought to our UI/UX department. But no active work as for now.

skotopes, 2024-07-07:
File on sd card with your information is the best option at this point. We are not planning to add additional text on lock screen

closed 2024-07-07 by skotopes