flipperdevices/flipperzero-firmware #2245: Weather Station - TFA/Dostmann protocol

buubsi, 2023-01-04:
### Describe the enhancement you're suggesting.

I found out the calculation for the temperature reading from a TFA/Dostmann 30.3121 outside wireless thermometer but i have not the coding skills to implement it myself.

The data is sent 2 times with a 30.000us delay inbetween, 1 = 550us, 0 = 1300us, gap = 1000us

Temperatur range is: -39,9°C to 59,9 °C
Frequency: 433.885 Mhz

Here is the Raw sub file for -19.4°C:

[-19.4.sub.txt](https://github.com/flipperdevices/flipperzero-firmware/files/10347557/-19.4.sub.txt)

------( sender id )--------( ? )---( a )----( b )----(c)-----(a)-----(b)--(checksum?)
[0000 1010 0000 1001] [1110] [0011] [0000] [0110] [0011] [0000] [1101]
---0----10----0-----9-----14-----3-------0------6------3-------0-----13

The calculation:

-AB-C
-19 .4°C

When a is the Number 5 or higher, then is the number positiv.

A = a - 5
B = b
C = c

When a is the Number 4 or lower, then is the number negative (and the calculation changes).

A = a - 4 ( and the number is negative)
B = b - 9
C = c - 10

Some examples are here:

[Samples.txt](https://github.com/flipperdevices/flipperzero-firmware/files/10347497/Samples.txt)

This is the transmitter:

[image]

Skorpionm, 2023-01-09:
record a few more pulses in RAW, without clipping the redundant data, and show what is really displayed on the receiver. maybe there is some moisture in there?

Skorpionm, 2023-01-09:
the data under a question mark is the fastest channel number and transmitter battery power. Do you happen to have a channel switch under the cover in your transmitter?

buubsi, 2023-01-09:
Here are the .sub files:

[sub.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10372828/sub.zip)

I dont think there is a moisture sensor in there, here is a picture of the receiver unit:
[image]
[image]

It can only display:
1 2 3 MIN MAX and the "wavething" on the left apperas when it waits for a signal (this is every minute).
Also when the battery runs out, it just stops working, there is no indicator for that (not implemented).

Also there are no channel switches. Maybe the id changes when the sender is reset.
I try this later.
[image]

buubsi, 2023-01-09:
Here are some more .sub files after the sender reset and indeed the bits 13-16 (id) changed.

[sub2.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10376489/sub2.zip)

Skorpionm, 2023-01-10:
Thanks, I'll definitely look, but I'm sorry now I'm sick

Skorpionm, 2023-01-11:
check https://github.com/flipperdevices/flipperzero-firmware/pull/2292

buubsi, 2023-01-11:
Nice Work @Skorpionm

[image]
[image]

closed 2023-01-11 by buubsi