flipperdevices/flipperzero-firmware #2246: Crash after dismissing FW Update Success screen using qFlipper controls.

ESurge, 2023-01-05:
### Describe the bug.

Crash with "NULL pointer dereference" after dismissing FW Update Success screen using qFlipper controls.

### Reproduction

- Use qFlipper or Updater app on FlipperZero to update to any firmware. (Tested with 0.74.0, 0.74.1, 0.74.2, and latest dev 72dc8e95)
- After update is complete, do not dismiss FW Update Success splash screen.
- Launch qFlipper and switch to remote control view.
- Press Ok button in qFlipper to dismiss FW Update Success splash screen.
- Notice FlipperZero crash with NULL dereference pointer.

### Target

FlipperZero

### Anything else?

- Crash does not occur if pressing Right button in qFlipper.
- Crash does not occur if pressing physical buttons on FlipperZero device.

skotopes, 2023-01-08:
Will be fixed in next release.

skotopes, 2023-01-09:
Fixed in #2280