flipperdevices/flipperzero-firmware #2247: Subghz remote button menu

fiveseven808, 2023-01-05:
### Describe the enhancement you're suggesting.

Much like how I can open an IR file and get a nice multiple line menu that lets me control multiple functions of a device (i.e. a fan), I propose a GUI menu like format similar to the IR file, where I can have multiple selectable transmissions in a *.sub file.

While we're at it, can we have also implement a dual row button system like the universal IR remotes? I get that you wont have space to but a word, or phrase or whatever, but what about putting 1-2 ASCII configurable characters in each "button"? and/or also allow the re-use of the existing universal layout button icons? 

### Anything else?

Use case: I have a bunch of these https://www.amazon.com/Etekcity-Wireless-Control-Learning-5Rx-2Tx/dp/B00DQELHBS lying around. (A bit unrelated, but subghz -> read does not reliable seem to be able to record the remote control transmissions but read RAW does)

As you can see, there is a different signal sent for each of the 8 channels. There are 2 signals for each channel, an on and and off. 

That's 16 different files in my flipper to full control all of these 8 outlets. The interface to activate each of these files is clunky and requires way more presses than I'd like. 

Yes, I know the purpose of this device is not designed to replace any sort of universal remote or application built device, but little quality of life improvements like making easily designable remote layouts would be cool. 

Instead of having a browser, and navigating to a directory... what if we could write a menu file that would be look like an IR universal remote when opened (i.e. the music remote0, but each of 8 on-screen GUI buttons could activate any specific file? 2 could open IR remote files, 2 could be subghz files (or subghz remote button menus even!), and 1 could open up an application, 1 could emulate a saved RFID file,  1 could be a favorite bad-usb script, and 1 could be a shortcut to the ibutton menu or something. 

Props to the flipper team by making a very very well put together piece of kit, both hardware and software wise. The linear menu system they included, most definitely works, but when a clicky dpad is all you have (vs say, a scroll wheel, or ipod ring lol) a more graphical non-linear menu could speed up usability (i.e. 2 columns vs 1 long column).

atsonkov, 2023-01-05:
Some custom firmwares have similar application, allowing you to configure each of the direction buttons and the OK button to send different SubGHz signals. It is called "Sub-GHz Remote".

This allows you to send 5 different signals without leaving the screen. Maybe it can be extended to allow up to 10 signals (by differing between short and long press).

[image]

See https://github.com/DarkFlippers/unleashed-firmware/tree/dev/applications/main/unirfremix and https://github.com/RogueMaster/flipperzero-firmware-wPlugins/blob/420/documentation/SubGHzRemotePlugin.md for some ideas.

I like the idea about being able to quickly navigate to more choices with the pad... maybe a label on the top of the screen showing the current choice, and 3 rows by 5? columns of possible selections will allow you to select lots of different things with low number of clicks. I even think it might be a nice interface to assign freely all kinds of things (e.g. assign IR signal to one button, SubGHz to another and BadUSB Script to third if you want).

fiveseven808, 2023-01-06:
This is super cool! I've been staying away from custom firmwares, but
seeing as how I use SubGhz a lot, I'll give it a shot. I noticed that apps
from unleashed run at a different API level from the official firmware (or
so the error I get tells me).

A bit off topic, but Is there any way to sort all of that out beforehand?
Or if I develop this unirfremix app to somehow assign multiple types of
actions (ir/badusb) to each dpad button, can I recompile it for the
official firmware?

skotopes, 2023-01-08:
Sounds like idea for an app. Not planning to add something like that to firmware itself.

closed 2023-01-08 by skotopes