flipperdevices/flipperzero-firmware #2253: Failed to build on Windows 10

cake-research, 2023-01-05:
### Describe the bug.

Failed to build firmware on Windows 10.
When run `fbt.cmd` it always try to download and unpack toolchain.
After that it run python and break with error: 
init_fs_encoding: failed to get the Python codec of the filesystem encoding...

### Reproduction

clone repository (dev branche)
run fbt.cmd

### Anything else?

In line #19 of file `scripts\toolchain\fbtenv.cmd`
`FBT_TOOLCHAIN_ROOT` variable set to `%FBT_ROOT%\toolchain\x86_64-windows`
It's folder uses to check toolset already installed and to set path variable.
But in line #10 of file `scripts\toolchain\windows-toolchain-download.ps1`
`$toolchain_dist_folder` set to `gcc-arm-none-eabi-10.3-x86_64-windows-flipper`
and uses to unpack toolchain.

For resolve this bug need to change `FBT_TOOLCHAIN_ROOT` variable in file `scripts\toolchain\fbtenv.cmd`
to `%FBT_ROOT%\gcc-arm-none-eabi-10.3-x86_64-windows-flipper`

hedger, 2023-01-08:
Your fix is incorrect, gcc-arm-none-eabi-10.3-x86_64-windows-flipper is a temporary folder and is removed after unpacking the toolchain.

I think your issue is related to path length limit on Windows. What is the full path to firmware's directory on your system? 

Try following [these steps](https://knowledge.autodesk.com/support/autocad/learn-explore/caas/sfdcarticles/sfdcarticles/The-Windows-10-default-path-length-limitation-MAX-PATH-is-256-characters.html#:~:text=Go%20to%20Windows%20Start%20and,Select%20the%20entry%20named%3A%20LongPathsEnabled.) to disable the limit, restart your system and try again.

hedger, 2023-01-08:
Just tried building a clean clone of the repo on WIndows 10, Version 10.0.19045.2364, and it succeeded. 
Reopen the issue if the path length fix doesn't work for you.

closed 2023-01-08 by hedger