flipperdevices/flipperzero-firmware #2254: Ability to run subghz protocol parser on sub file outside Flipper

exander77, 2023-01-05:
### Description of the feature you're suggesting.

I would like to improve some flipper modules and either I am missing something or it is not possible, but I would expect that it is way easier to run protocol parser outside Flipper on recorded sub files.

fiveseven808, 2023-01-06:
What I want, is the ability to run subghz protocol parser on already recorded "read-raw" files! 

I think it would be really useful to be able to record a signal, run the parser, and have the parser automatically spit out the results for the various supported modulations. That way, someone can get a feel on what an unknown signal would look like if de-modulated in any of the supported modulations, and can maybe better pick the correct one for the "read" function.

exander77, 2023-01-06:
I think, you can do this using cli.

skotopes, 2023-01-08:
Yep there is a CLI tool for RAW file decoding. As for GUI - best if it will be external app.

closed 2023-01-08 by skotopes