flipperdevices/flipperzero-firmware #2260: Raw data in .sub-files doesn't always start with a positive integer

bartbroere, 2023-01-06:
### Describe the bug.

When dumping some `.sub` files using the Sub-GHz > Read Raw function I noticed that sometimes the `Raw_Data` lines don't start with a positive number, and sometimes they start with two positive numbers.

When reading in the line with Python, this can easily be fixed with something like this, after splitting the line on spaces:
```python
if data[0] > 0 and data[1] > 0:
    data = data[2:]
if data[0] < 0:
    data = data[1:]
```

However, the documentation of the file format mentions that the values must start with a positive number:
> `RAW_Data`, contains an array of timings, specified in micro seconds. Values must be non-zero, start with a positive number, and interleaved (change sign with each value). Up to 512 values per line. Can be specified multiple times to store multiple lines of data.

Perhaps this can be fixed either in the documentation or in the firmware. I'm happy to help with a pull request if this is indeed a bug (or documentation error).

### Reproduction

Sample file:

```
Filetype: Flipper SubGhz RAW File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: RAW
RAW_Data: 63293 18033 -19558 659 -1458 119225 ...
RAW_Data: -18560 295 -558 295 -598 577 ...
```

I have truncated the lines, but the pattern continues with the alternating sign.

skotopes, 2023-01-06:
Looks like some incorrectly handled counter overflow. I'll check our code.

skotopes, 2023-01-06:
Can you provide full file and firmware version that was used to record it?

bartbroere, 2023-01-06:
Last time I checked I was running the latest stable firmware, somewhere earlier this week. I'm on my phone now, so I can't find the exact version immediately.

I'll do that as soon as possible.

This is the best way I can share the entire file from my phone:

```
Filetype: Flipper SubGhz RAW File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: RAW
RAW_Data: 158 33827 -10680 165 -464 197 -432 1589 -266 97 -366 163 -66 187459 -12300 725 -398 163 -132 227 -100 261 -66 229 -98 565 -100 99 -198 1785 -12216 131 -264 65 -100 131 -364 65 -198 427 -164 97 -132 131 -164 163 -66 65 -66 263 -64 97 -198 87831 -11282 99 -656 463 -496 97 -332 165 -198 297 -232 231 -166 67 -132 99 -132 41853 -12092 165 -100 263 -66 131 -360 197 -394 65 -264 229 -134 65 -166 429 -200 229 -66 129 -298 97 -262 163 -66 1249 -98 47725 -9706 99 -262 133 -232 99 -168 265 -166 65 -132 131 -68 197 -134 295 -98 529 -98 361 -100 327 -130 97 -66 1581 -10400 97 -568 821 -660 65 -498 263 -100 97 -198 131 -164 263 -100 361 -98 261 -164 30757 -12342 65 -164 65 -196 757 -362 97 -100 97 -230 165 -296 131 -230 263 -330 163 -66 65 -130 917 -100 1117 -12154 133 -66 197 -132 199 -100 467 -132 67 -266 297 -198 131 -100 131 -100 265 -132 20109 -10106 65 -1514 195 -166 65 -100 97 -1328 495 -200 231 -68 495 -200 65 -596 1985 -594 621 -264 629 -310 583 -314 275 -600 603 -318 597 -288 283 -606 609 -302 595 -304 315 -574 605 -296 305 -618 279 -600 327 -576 317 -600 287 -614 585 -308 595 -296 603 -318 285 -600 617 -292 597 -30928 337 -552 633 -288 605 -284 617 -314 287 -592 621 -284 585 -316 323 -562 615 -314 583 -318 287 -600 613 -276 321 -584 317 -568 321 -608 319 -560 313 -600 611 -314 589 -284 633 -276 325 -558 627 -284 619 -30908 355 -560 631 -258 633 -284 619 -278 321 -596 583 -314 593 -316 287 -600 613 -276 619 -284 323 -570 611 -304 317 -576 297 -594 343 -584 277 -598 329 -578 611 -302 593 -304 615 -284 307 -598 599 -318 601 -30908 317 -624 547 -376 525 -362 565 -348 243 -622 583 -330 569 -348 247 -634 581 -336 577 -316 281 -600 579 -342 281 -622 275 -598 289 -618 317 -592 273 -630 583 -312 589 -314 599 -290 283 -638 575 -310 593 -30946 315 -620 575 -342 557 -328 567 -350 243 -634 581 -336 543 -348 281 -598 581 -340 577 -316 281 -598 581 -342 281 -622 277 -596 291 -620 315 -578 309 -594 589 -314 579 -316 599 -302 315 -588 601 -294 601 -30936 329 -580 615 -312 595 -290 595 -318 287 -598 617 -290 595 -318 287 -598 587 -292 621 -318 287 -598 589 -290 323 -612 283 -596 307 -594 293 -606 319 -594 589 -290 619 -318 593 -312 287 -594 589 -314 581 -30944 313 -584 603 -292 601 -316 583 -316 321 -564 613 -312 585 -318 287 -568 609 -306 617 -284 323 -570 609 -304 313 -586 311 -600 295
RAW_Data: -578 317 -600 287 -614 613 -276 593 -298 603 -320 273 -598 625 -284 605 -30884 387 -558 585 -316 581 -316 601 -288 315 -576 611 -312 593 -294 297 -584 609 -310 593 -296 301 -622 575 -310 305 -600 311 -592 317 -564 309 -594 293 -606 611 -314 593 -290 593 -320 289 -578 617 -292 599 -30906 327 -580 615 -312 593 -294 593 -318 285 -598 617 -292 591 -318 287 -606 605 -300 577 -318 321 -570 613 -300 311 -590 311 -566 321 -592 317 -600 285 -612 579 -310 597 -294 601 -318 277 -632 591 -318 593 -30914 321 -598 591 -290 611 -284 617 -312 287 -596 587 -316 583 -316 323 -564 613 -312 591 -282 323 -570 609 -306 319 -592 319 -570 293 -602 313 -596 315 -566 611 -306 617 -284 607 -310 271 -604 613 -318 599 -30942 319 -586 613 -280 629 -258 621 -320 289 -580 615 -294 599 -320 287 -580 617 -292 599 -318 289 -602 587 -292 321 -578 317 -598 311 -594 291 -604 317 -598 589 -290 613 -318 581 -312 287 -594 621 -284 593 -30952 323 -596 593 -290 609 -284 619 -278 321 -596 587 -316 593 -316 287 -600 611 -312 589 -282 325 -570 611 -306 317 -590 319 -564 309 -594 329 -576 319 -578 617 -292 599 -320 579 -312 321 -558 623 -286 617 -30876 343 -592 609 -276 591 -298 605 -318 289 -602 587 -292 619 -320 287 -600 591 -290 613 -284 325 -598 591 -290 317 -576 319 -596 311 -592 291 -600 317 -598 591 -288 613 -284 631 -274 307 -598 611 -286 597 -30890 321 -610 615 -280 597 -290 617 -284 327 -600 591 -290 609 -284 323 -600 591 -288 611 -284 323 -580 603 -294 301 -618 283 -600 309 -594 291 -604 319 -596 589 -290 615 -284 617 -314 287 -594 585 -316 585 -30918 313 -592 605 -266 625 -314 583 -280 309 -600 611 -284 627 -258 323 -580 635 -258 635 -286 309 -600 583 -314 279 -598 321 -590 319 -566 319 -606 319 -568 605 -302 611 -286 599 -288 317 -574 609 -286 627 -30870 291 -598 615 -318 603 -258 633 -284 323 -564 623 -308 591 -300 311 -586 603 -258 637 -284 309 -600 585 -316 277 -600 321 -588 319 -566 319 -606 319 -568 605 -300 613 -286 597 -288 315 -578 613 -300 595 -30890 317 -584 623 -280 597 -290 615 -320 287 -602 589 -288 613 -284 325 -600 591 -288 611 -284 325 -600 591 -288 315 -576 319 -578 345 -558 329 -576 319 -598 591 -288 613 -284 633 -280 325 -562 613 -312 583 -30874 357 -570 615 -298 573 -318 599 -288 315 -574 609 -286 629 -310 285 -606 573 -328 575 -318 275 -630 595 -318 273 -600 293 -602 317 -594 311 -594 289 -598 621
RAW_Data: -278 597 -292 617 -318 289 -578 617 -292 595 -142568 97 -466 131 -632 1293 -98 67 -132 331 -166 427 -66 297 -230 461 -100 197 -100 99 -66 18349 -10428 65 -166 131 -1260 599 -232 165 -100 365 -262 199 -168 99 -134 1353 -100 128837 -12414 165 -98 131 -228 65 -166 65 -926 97 -332 97 -364 165 -198 67 -166 133 -66 131 -12104 229 -852 133 -496 165 -300 131 -530 133 -232 299 -132 65 -132 34439 -12080 97 -302 231 -132 165 -360 99 -556 99 -98 97 -264 97 -862 [... cut ...] 1 -266 1163 -66 36719 -11834 99 -232 429 -132 131 -98 99 -198 265 -400 165 -362 65 -100 395 -130 593 -98 131 -132 5841 -11214 163 -100 65 -100 199 -566 131 -132 201 -134 131 -66 65 -132 65 -532 563 -238 21839 -12162 67 -266 131 -66 233 -100 297 -98 131 -230 557 -328 131 -164 65 -560 327 -66 89251 -12038 63 -98 99 -164 65 -98 163 -132 65 -132 263 -98 427 -98 229 -166 689 -132 955 -100 6323 -10898 129 -658 427 -166 65 -232 65 -100 99 -196 65 -396 293 -890 791 -98 401 -66 8811 -10702 99 -862 65
```

bartbroere, 2023-01-06:
I notice that it is not the file from the first comment, but it does have the same issue.

skotopes, 2023-01-06:
Yep, I've reproduced issues. We'll address it in next release.

skotopes, 2023-02-10:
We've added couple fixes in 0.77. Please reopen if issue is still persist

closed 2023-02-10 by skotopes