flipperdevices/flipperzero-firmware #2265: Time to poweroff the device

alvarotorijano, 2023-01-07:
### Describe the enhancement you're suggesting.

Hi, the time I have to keep pressed the back button to poweroff the device is excevely long. Would be worth it to lower the time or add a way to customize this time (in a menu or a config file)

skotopes, 2023-01-08:
Do you use this feature often? What's your typical use case for the device?

alvarotorijano, 2023-01-08:
Now I'm still learning to use it so there are a lot of times when I power up the device try something and then power it off. It will be also useful when I only have to emulate a known remote or an access card and then power it off again

skotopes, 2023-02-10:
We've discussed it internally and would like to keep this button behavior.

closed 2023-02-10 by skotopes

alvarotorijano, 2023-02-10:
well, it is ok anyway thanks for the time and dedication. It seems that after a few updates, the device was running slower than it should have. I discovered it after a bad update and using the DFU mode. No the time I have to keep pressing the button is clearly smaller.
Anyway, I only wanted to be able to change it which can be solved just... Writting a fap!