flipperdevices/flipperzero-firmware #2272: U2F doesn't work over Lightning cable in iOS

TheXHalferX, 2023-01-07:
### Describe the bug.

Standard behaviour: LED is green/red, text on screen is "Connected" using Flipper's U2F on PC, Mac, iPad (over USB-С)
Connecting Flipper in U2F over Lightning to iPhone does nothing ("Connect me...")

### Reproduction

1. Launch U2F
2. Head over to any website with U2F support (i.e. GitHub.com)
3. Set up U2F key (if not already)
4. Try to login on iPhone connecting Flipper to iPhone over Lightning

### Target

U2F Plugin

### Anything else?

Logs were not generated, so issue may be with Lightning interface of with iPhone (all of them) itself

ankris812, 2023-01-08:
there could be various things make it fail. interface itself. iphone not detecting it, tha cable. what cable are you using?

TheXHalferX, 2023-01-08:
I used original Lighting to USB-C, Lightning to USB-A + USB-C/USB-A adapter, used same but from third party manufacturer. I'll investigate more tomorrow. 

Could the issue be because of Flipper not detecting iPhone/interface?

ankris812, 2023-01-08:
> I used original Lighting to USB-C, Lightning to USB-A + USB-C/USB-A adapter, used same but from third party manufacturer. I'll investigate more tomorrow.
> 
> Could the issue be because of Flipper not detecting iPhone/interface?

nope. its because iphone not detecting it. iphone should detect it as usb accesory, but looks like its not doing it

TheXHalferX, 2023-01-08:
I see. I'll try use other cables and report back here.

skotopes, 2023-01-08:
Flipper conforms to TYPE-C/USB2.0 specification. So in theory it should work with any host that supports this combination.

Apple devices with type-C connector are working with flipper correctly. Not everything is supported on IOS/Application side, but at least it's properly detected and powered.

As for devices with lightning connector: they are in the grey zone, lightning - type-C cable is not what you think it is and it's not going to work with flipper or any other device that uses legacy type-c connection schema.

There is an OTG adapter that may work but no guarantees.

And there is nothing can be done on flipper side.

closed 2023-01-08 by skotopes

TheXHalferX, 2023-01-08:
Didn't know that! Thank you for the information!

KA19XFinntage, 2024-09-03:
I tried the otg adapter but it did not work it said the flipper consumed to much power