flipperdevices/flipperzero-firmware #2274: EM4305 write support

Bon-Jorik, 2023-01-08:
### Description of the feature you're suggesting.

There is still no way to write something on EM4305 chips using flipper. I think, adding this option will be very helpful cause of prevalence of this model.

Best regards ^_^

### Anything else?

I can send a raw data from chip which i cant write if it needed

skotopes, 2023-01-08:
Yep, we have it in our backlog. Planning to implement soon.

Bon-Jorik, 2023-04-08:
Is there anything new? I have an impant with this chip and still cant write it with flipper. Very sad to see that no one is interested

ax-initrd, 2023-05-04:
Hope this gets implemented soon. Most sources now sell EM4305 instead of EM41000 so this would be greatly appreciated. Thank you!

ololonly, 2023-05-13:
strongly waiting this feature too

hlakkerderie, 2023-06-23:
I'm also waiting for this. 
Offcourse it will be possible (I think), but we have to wait for the smart guy's to program it right.

hlakkerderie, 2023-08-25:
For people that are searching for writable tags that works with Flipper,  I ordered it by this guy. 

https://www.ebay.com/itm/272580275835

A ordered by others a couple times, but get every time the wrong tags.

CDarbonne, 2023-09-20:
Very much waiting for this as well. If you don't mind me asking, what's the priority on this? Seems like it's been a while since "soon" and just wondering if it's just not high priority or if it's presenting more problems than anticipated. Thanks!

skotopes, 2023-09-20:
After we merge new NFC stack in 0.93~0.94

Polsaker, 2023-11-06:
any news?  This would be a really cool feature

awesomesk1ll, 2023-11-07:
up

masterxsoft, 2023-11-19:
I'm also waiting for support, I have a lot of EM4305 lying here

gztproject, 2023-12-13:
> After we merge new NFC stack in 0.93~0.94

Any news on this?

Korol-tvoego-serde4ka, 2024-01-29:
> After we merge new NFC stack in 0.93~0.94

Any news on this?

RaphaelNeumann, 2024-02-29:
> I'm also waiting for support, I have a lot of EM4305 lying here

Any news on this?

RaphaelNeumann, 2024-02-29:
> After we merge new NFC stack in 0.93~0.94

Any news on this?

skotopes, 2024-03-12:
Nope, no news.

This task is currently in backlog, we'll get back to it at some point, but no specific timeline. There is a chance that someone from community will implement it first.

blivessa, 2024-03-24:
> Nope, no news.
> 
> 
> 
> This task is currently in backlog, we'll get back to it at some point, but no specific timeline. There is a chance that someone from community will implement it first.

Is there a branch open for this?

skotopes, 2024-03-25:
> > Nope, no news.
> > This task is currently in backlog, we'll get back to it at some point, but no specific timeline. There is a chance that someone from community will implement it first.
> 
> Is there a branch open for this?

None that I'm aware of

aaronparres, 2024-04-18:
Would really appreciate to have this write option, my kyber crystals are waiting 😄

Mahad444, 2024-06-23:
Can any one show me on how to write or duplicate EM4100,or is this type read-only

ololonly, 2024-06-24:
> Can any one show me on how to write or duplicate EM4100,or is this type read-only

@Mahad444 at this moment you should use RFID with T5577 chip, it's rewritable and can emulate EM4100

rodry7, 2024-09-25:
Any news on this? does it come with new flipper version 1.0?

vertexgamer, 2024-10-02:
Almost 2 years and still no support..... Guys if you don't want to implement it just say it! it's ok. But don't waste people time with "coming soon" or "we are working on this" or "it's in our backlog"

skotopes, 2024-10-02:
@vertexgamer you are welcome to implement it. Meanwhile it is in backlog with lowest priority possible.

iso250, 2024-10-20:
> Yep, we have it in our backlog. Planning to implement soon.

Has soon arrived yet?  If you're not going to do it I'd rather you just say so.

skotopes, 2024-10-20:
> > Yep, we have it in our backlog. Planning to implement soon.
> 
> Has soon arrived yet? If you're not going to do it I'd rather you just say so.

Where are you guys coming from? Answer is literally in previous message: there is no active work done, you are free to implement it.

iso250, 2024-10-20:
> > > Yep, we have it in our backlog. Planning to implement soon.
> > 
> > 
> > Has soon arrived yet? If you're not going to do it I'd rather you just say so.
> 
> Where are you guys coming from? Answer is literally in previous message: there is no active work done, you are free to implement it.

Where are we coming from? From a place where we have a device that’s supposed to write to RFID cards, yet we bag only wire to a subsection of them and our requests for over 2 years are just being ignored. That’s where we’re coming from. 

Implement it myself?  I BOUGHT THIS DEVICE SO IT COULD DO THE TASK.

skotopes, 2024-10-20:
@iso250 4305 were never announced to be supported, only 5577.

` I BOUGHT THIS DEVICE SO IT COULD DO THE TASK.`
and in a small letters `but never read device specification`

iso250, 2024-10-20:
> @iso250 4305 were never announced to be supported, only 5577.
> 
> ` I BOUGHT THIS DEVICE SO IT COULD DO THE TASK.` and in a small letters `but never read device specification`

“Out of the box, Flipper Zero can emulate remotes for popular garage doors and barriers.”

This was is regular letters on the Kickstarter page. That never happened. 

Like so many Kickstarter projects this had so much promise but turned out to be such a pos.

Astrrra, 2025-01-16:
Will be fixed by #4069

PR #4069 (2025-02-12): [FL-2754, FL-3945] EM4305 support
# What's new

- Internal support for writing to EM4305 blanks
- Support for writing EM4100, Electra, Viking, and Gallagher data to EM4305 blanks
- This PR does not introduce support for writing other data formats to EM4305 blanks yet, it only lays the internal groundwork, other formats can now be added on an individual basis.
- Fix #2274

# Verification 

- Try writing existing data formats to a T5577 tag, check that everything works as it did before.
- Try writing EM4100/Electra/Viking/Gallagher dumps to a EM4305 tag, check that everything gets written and re [... cut]
closed 2025-02-12 by skotopes